andreknieriem/photobooth #101: Kamera löst nur via Konsole aus..

UncleCCC, 2019-10-04:
Mahlzeit..
Ich hab Photobooth frisch auf meinem NucI3 installiert, 
Als OS ist MX Linux am werkeln (Basiert auf Debian)
Die Installation verlief ohne Probleme, jedoch wird die Kamera übers WebIf nicht ausgelöst, immer wieder der Hinweis nach dem Countdown "Es ist ein Fehler aufgetreten..."
 Direkt über die Konsole funktioniert das ganze ohne Probleme..
$ env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download
New file is in location /capt0000.jpg on the camera                            
File capt0000.jpg exists. Overwrite? [y|n] y   
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera

andi34, 2019-10-04:
Camera?
Camera set to jpeg only?

You could try updated installation instructions and see if it's an permissions issue.
https://github.com/andreknieriem/photobooth/pull/86/files

UncleCCC, 2019-10-04:
Ah, after setting the rights it works, but only 2 photos.
Also in the console it does not work anymore.
Then the following error:
*** Error ***
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***
Probably a bug on GPhoto?

andi34, 2019-10-04:
Could be.
Could also be interesting for you:
https://github.com/andreknieriem/photobooth/issues/73
https://github.com/andreknieriem/photobooth/issues/93

plin2, 2019-10-04:
I would go for the gvfs service (#93). Disabling it solved the problem for me.

UncleCCC, 2019-10-05:
You had the same problem?

Don't understand, which memory should I hang from?
Just have sda, that's my hard drive...

plin2, 2019-10-05:
Well, when I googled the error message I found a hint that gvfs-gphoto2-volume-monitor ist blocking the camera. The solution in that post was to hinder the daemon from execution:
chmod -x /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
That solved the situation for me.

UncleCCC, 2019-10-05:
The problem's killing me.
Does not exist...
$ find gvfs-gphoto2-volume-monitor
find: ‘gvfs-gphoto2-volume-monitor’: Datei oder Verzeichnis nicht gefunden

5 - 10 photos... then the error

*** Error ***
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') *

andi34, 2019-10-05:
The hard way is documented in the README and Wiki:

Ensure that the camera trigger works:
```
sudo rm /usr/share/dbus-1/services/org.gtk.vfs.GPhoto2VolumeMonitor.service
sudo rm /usr/share/gvfs/mounts/gphoto2.mount
sudo rm /usr/share/gvfs/remote-volume-monitors/gphoto2.monitor
sudo rm /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
```

andi34, 2019-10-05:
From https://github.com/andreknieriem/photobooth/issues/73 as solution:

> Habe jetzt auf "jpg-only" gestellt und die Größe verringert.

Maybe lowering photo resolution a little bit helps?

andi34, 2019-10-05:
From looking at issues in gphoto at GitHub a problem also seems to be the focus on Canon sometimes:

> I find out that if I force the focus with the remote just before doing the --capture-image-and-download, it works perfectly.

You might also take a look here https://github.com/gphoto/libgphoto2/issues if none of the possible solutions mentioned here works.

plin2, 2019-10-05:
I think during my first installation of photobooth I had the problem that the user that's running the apache didn't create a .gphoto2 directory in it's home directory. I started with
ps -ef | grep apache
that showed me www-data is the owner of the process
`grep www-data /etc/passwd`
showed me
`www-data:x:33:33:www-data:/var/www:/usr/sbin/nologin`
I changed /usr/sbin/nologin to /bin/bash, that allowed me to su into that user.
after `su - www-data` I tried to take a picture, which didn't work.
Finally I created the directory .gphoto2 to enable gphoto to store it's settings

Well, that was the first photobooth installation, for the second one it wasn't necessary. Probably some ownerships/permissions were not set correctly.

andi34, 2019-10-05:
@plin2 if he says 5-10 Photos work until it crashes I'd say it's not an permissions issue.
Gphoto2 issue page contains a lot of such issues with canon cameras, on some issue threads user tell how they work around the issue.

devnull1975, 2019-10-05:
Also meine Kamera löst nur aus wenn ich die Speicherkarte entferne und im Kamaramenü einstelle das die Kamera auch ohne Speicherkarte auslösen soll. 
Für mich ist das ok so.

andi34, 2019-10-05:
There's different options you could try on gphoto2 too:
```
--keep-raw
```
Or setting a capture target on your camera:
```
--set-config capturetarget=x
```
Explanation: https://github.com/andreknieriem/photobooth/wiki/gphoto2

devnull1975, 2019-10-05:
@andi34 i know this solutions but i dont need RAWs and a copy of the images too.
My System works fine and i never change a running system.
Maybe it can be the solutions auf the isuess the the PI only take a Couple of pictures.
Also it can be that the wire from the PI to the cam make some Problems. 
Is it the original Wire or a cheaper from China?

sualko, 2019-10-06:
> Gphoto2 issue page contains a lot of such issues with canon cameras, on some issue threads user tell how they work around the issue.

Therefore I think this is not an issue with photobooth and therefore I close this issue, but please don't hesitate to continue your discussion.

closed 2019-10-06 by sualko

andi34, 2019-10-07:
You've removed the files on the initial setup already as mentioned in the README?
In that case they can't exist anymore.

You could also try using latest version instead latest stable version of gphoto2. Follow the instructions to install gphoto2 again.

andi34, 2019-10-07:
You've tried lowering the photo resolution on your camera?

UncleCCC, 2019-10-07:
Yes, I have tested 1920 x 1280, and the same problem...
Gphoto 2 I have already updated to the latest version, problem persists.

I remembered that 2 weeks ago I installed Photobooth on a Raspberry without GUI. I couldn't find the problem (Could have been a coincidence, I took only a few pictures) Now I installed it on Intel Nuc with Linux MX and Xfce. Could the problem be?

I install again on the Raspberry Photobooth

Translated with www.DeepL.com/Translator

andi34, 2019-10-07:
> From looking at issues in gphoto at GitHub a problem also seems to be the focus on Canon sometimes:
> 
> > I find out that if I force the focus with the remote just before doing the --capture-image-and-download, it works perfectly.
> 

Hatte ich schon erwähnt ;)
Aber dann ist ja im, wenn du das Problem gefunden hast. 
Kannst ja gerne https://github.com/gphoto/libgphoto2/issues ein issue aufmachen bzw deine Erfahrungen bei einem passenden Issue mitteilen in der Hoffnung daß es upstream gefixt wird.

UncleCCC, 2019-10-07:
Es gibt einige Issuses...
Die aber nie wirklich komplett gelöst werden, habe auch am Freitag ein neues erstellt, wurde bis heute nicht freischaltet..
Denke werde mir eine andere Cam holen.. 
Schade eigtl..

andi34, 2019-10-07:
Wenn du Fokus manuell einstellst? Oder vorher per remote antriggerst?
Es werden meistens issues nicht am selben Tag behandelt. OpenSource lebt durch die Community. Head devs haben in aller Regel auch noch reguläre Jobs denen sie nachkommen. Mal hat man mehr Zeit, mal weniger, um sich um Probleme und Anfragen zu kümmern.

Wie schaut's aus mit dem Python workaround? https://github.com/gphoto/libgphoto2/issues/30#issuecomment-346974478

UncleCCC, 2019-10-07:
Das soll auch überhaupt kein Vorworf an den/die Devs gewesen sein. 
Und ist ganz klar das bestimmt andere sachen wichtiger sind und Vorrangig sind :)

Dieses Workarounf habe ich gesehen, aber erstens habe ich wirklich wenig Ahnung davon, und weiß auch nicht genau was ich damit überhaupt anstelle, anscheinend den Autofokus ganz deaktivieren?
Aber ich denke der Autofokus ist sehr nützlich, gerade bei automatischen Aufnahmen, denke die Bilder werden dann wesentlich besser als Manuell..

msmedien, 2019-10-10:
@UncleCCC Bei meiner Kamera (Sony alpha58) muss ich den Autofokus komplett ausschalten, da dieser sonst, vorallem bei schlechteren Lichtbedingungen, anfängt zu pumpen. Da die Software in einem gewissen Zeitrahmen eine Antwort erwartet, die Kamera aber noch versucht zu fokusieren, wird das als Fehlerhaft gewertet...

Um trotzdem Fotos mit genügend Tiefenschärfe zu machen (sollten sich die Personen einmal etwas weiter vorn oder hinten befinden), kannst Du doch die Blende an deiner Kamera nach oben setzen. - Damit solltest Du genügend "Distanzpuffer" haben. Ein Blitz, egal ob intern oder extern, sollte Dir dann genügend Licht geben um die geschlossene Blende zu kompensieren.

Eine neue Kamera sollte daher nicht nötig sein. - Dort wäre das gleiche Problem mit der Fokusierung.