andreknieriem/photobooth #103: Success !! finished project  (nearly)

vdubuk, 2019-10-06:
Hi to all, I just wanted to share with everyone my finished project for a Wedding Postbox / Photo kiosk

[image]

[image]

Rather messy inside 
[image]

sualko, 2019-10-07:
Looks awesome :tada: :+1: Thanks for sharing. Hope you have a good time with your photobooth.

closed 2019-10-07 by sualko

andi34, 2019-10-07:
By @thymon13 :

> Hello,
> 
> I 'm glad to present you my Photobooth.
> Thanks to all the people who contribute to the improvement of this project !
> 
> I was inspired by other photobooth on the internet. And I liked the one that looks like Instagram.
> **So I started !!**
> 
> [image]
> [image]
> 
> **Here it is in action**
> 
> [image]
> [image]
> [image]
> 
> **Video test QR Code :**
> 
> https://youtu.be/EG5JD2_vvRg
> 
> **I recently added the print function. And I have edit the file "print.php" to add my own frame around the picture (before the last update on this project)**
> 
> [image]
> [image]
> 
> **The heart of my baby :) :)**
> 
> [image]

By @flighter18 

> @thymon13 that looks really amazing!
> great work!
> mine is not that professional but I really like my bluetooth buzzer :)
> [image]
> [image]
> 
> @andi34 let's start a wiki. that would be great

By @Nie-Oh 

> Hello,
> 
> Here are some Pictures of my Photobooth.
> 
> [image]
> 
> The Case is made of IKEA 1x1 Kallax. I use a Touchscreen Monitor directly connected to the Raspi.
> 
> [image]
> 
> Also, but i have no Images of these Setup, i have an 24" Medion All in One PC and a Mitsubishi 9550 Photoprinter. Every Minute i syncronize the Pictures with WINscp. The Customers can use these PC for printing the Pictures.....

andreknieriem, 2019-10-07:
So many cool Projects :) love them and thanks to all for your support on my initial project!