andreknieriem/photobooth #104: Feature request: Main Css Styling in Admin Panel

andreknieriem, 2019-10-08:
Hey everybody. I have a new idea for the photobooth. In some of my earlier projects I implemented a wordpress theme with color-picker options. I think this could be a nice feature.

Here is an image of the backend:
[image]

To not conflict the scss styling, maybe we can use this:
https://leafo.net/scssphp/docs/ it is a php scss compiler. So we can load the scss file and set the variables from the admin panel to generate the new css.

I have used https://www.eyecon.ro/colorpicker/ to pick the color. I like it. What are you thinking? Maybe with some drag/drop upload for background images?

andi34, 2019-10-08:
I think this is a good idea.
To change the background via admin panel was requested once (https://github.com/andreknieriem/photobooth/issues/57) which I added to the feature request collection issue/thread.

To not have to many conflicts later I'd wait for https://github.com/andreknieriem/photobooth/pull/90 to be merged, as the code is rewritten in many parts.

andi34, 2019-10-08:
Or maybe on top of https://github.com/sualko/photobooth/commits/feature/misc

sualko, 2019-10-08:
No need to use a php scss compiler, because feature/refactor uses css vars and feature/misc even supports custom colors from the config file. Therefore only the admin page has to be modified. I personally would prefer to use the [html5 input type color](https://developer.mozilla.org/en-US/docs/Web/HTML/Element/input/color), which has a pretty good [browser support](https://caniuse.com/#feat=input-color).

msmedien, 2019-10-16:
Is it possible to integrate an option for the font-size of the buttons? On a screen resolution of 1920x1080 is the text extremly small and i need to adjust it after every update manually in the css file. :/

Or is it possible to include the btn. class in this way, that for different screen resolution a different font-size is set as standard? :) Thanks a lot :)

sualko, 2019-10-16:
@msmedien please open another issue for your request, because otherwise can't organize all requests and the author of this issues gets notifications for your discussion.

andi34, 2020-01-13:
As information: I am working on it.

[image]

andreknieriem, 2020-01-13:
Looks very nice!

andi34, 2020-01-13:
My current WIP implementation would also fix https://github.com/andreknieriem/photobooth/issues/198 as the color for Cheese!, Countdown and the background can be adjusted seperately (same color, e.g. black = no text visible)

closed 2020-01-15 by andi34

flighter18, 2020-01-16:
Good morning @andi34 

Would it be a big task to distinguish font color between back-end and front-end?

If you choose white as text color, the front-end look nice:
[image]

But you can't read the font in the admin panel, if the panel color is white:
[image]

andi34, 2020-01-16:
@flighter18 you need to adjust the "admin and login panel color" if you like to use white text

andi34, 2020-01-16:
"Field color" maybe also

andi34, 2020-01-16:
Revert https://github.com/andreknieriem/photobooth/pull/218/commits/2a1da4e444ac962c22755cd1083849907c3ab1bc if you don't need/like to change the font color on start screen

andi34, 2020-01-16:
https://github.com/andi34/photobooth/commit/212d7a28ab42b966123cae8722dd0e7ea01c0ad8

I'll merge/commit that later here too.