andreknieriem/photobooth #105: Bug: No countdowns while collage-feature

msmedien, 2019-10-08:
Hi, if i use the collage-feature then appears the first countdown as always, but for the second, third und fourth photo is no countdown displayed... After the first photo it just says "processing collage" and it makes every 5 seconds a new photo... The guests dont know when the next photo will be taken... :( Any solutions for this problem? I think it's more a bug then a missing feature, isnt it?

andi34, 2019-10-08:
@sualko is working on that already. In future you have to take 4 pics if I have seen right, every pic has its own countdown then

andi34, 2019-10-08:
For reference https://github.com/sualko/photobooth/commit/559259429749e389c0c4d1611ed9f556c0212126

Will be added once the refactoring is done.

msmedien, 2019-10-09:
@andi34 @sualko I have to press 4 times the "take picture" button...? Sorry, but this is the biggest bullshit i've ever heard... No user of the photobooth will go 4 times to the screen for just 1 photo in the end. >:( Also with a trigger is this a bad solution.

One button with 4 countdowns/photos in a row... Maybe as an option in admin panel, where you can switch between 4 single photos or all in a row...?

sualko, 2019-10-09:
> Sorry, but this is the biggest bullshit i've ever heard...

This is an open source project and if you don't like the behavior, don't use the project, create a pull request, or ask nicely if one of the maintainers can implement your request in there free time.

andi34, 2019-10-09:
I am with @salko, we're all doing this without getting paid in our sparetime. The project is opensource and everyone can tell ideas or improvement etc...
And complaining about something which is not yet merged that hard? Wrong language to talk for OpenSource stuff

msmedien, 2019-10-10:
@andi34 @sualko I'm really sorry for the rude wording. I know that everyone give the best for this awesome project. But i can't understand why you try to make this feature much more complicated to the user as necessary... :/
Back to the main thread: The "original" collage feature is a good solution - only the countdowns between the photos are missing.

As allready said: Sorry for rude wording. :)

andi34, 2019-10-10:
@msmedien nobody is vindictive, but next time you might first think about needed words - such rude words could destroy an OpenSource project (in the past i've had to close threads in an android forum and keep source improvements for myself for a while because i thought some user don't deserve the work i am done in my sparetime).
We've been testing the refactoring and new features, you've been asked too to test.
Nothing was merged, so there was no need for the rude words.
You could have tested yourself and give proper feedback instead (e.g @flighter told in nice words he don't like it and i already told we can make it optional).

You can follow the installation instructions posted yesterday. Countdown works between every photo in collage.

tobiashaas, 2019-10-10:
Sometimes it is not bad if you have to press for each photo.  I once had a case where they wanted to replace their props after each photo.  this does not work if he independently takes all photos.

andi34, 2019-10-10:
There's now the option called "Take collage without interruption" to not always click "Take Pic".

If you've setup already from the instructions yesterday:
```
git fetch origin
git checkout origin/master
yarn install
yarn build
```

closed 2019-10-10 by sualko