andreknieriem/photobooth #116: v2.0.0: Gallery Buttons (Print/QRCode) doesn't work on IOS 12

flighter18, 2019-10-10:
If I open a pic in the gallery, the button in the right corner doesn't work except the close button. I uploaded a video:
https://youtu.be/e3e1k_bvFxc

closed 2019-10-11 by sualko

flighter18, 2019-10-11:
@sualko i checked out your fix:
```
commit b889e2da19ea30580c9e55f685fe8e7a98166b56 (HEAD)
Merge: 5cee070 331d1ef
Author: raphael.schib <raphael.schib@kara-network.ch>
Date:   Fri Oct 11 17:15:34 2019 +0100

    Merge commit '331d1ef4adeb4269893bb7bdc840fd0b2b1c6ec4' of https://github.com/andreknieriem/photobooth into HEAD

commit 331d1ef4adeb4269893bb7bdc840fd0b2b1c6ec4
Author: sualko <klaus@jsxc.org>
Date:   Fri Oct 11 15:15:14 2019 +0200

    fix gallery buttons on small screens

    fix #116
```
but unfortunately it still not working

andi34, 2019-10-11:
@flighter18  you need to run `yarn build` again.

Once something is in master branch merged it's enough to `git fetch origin && git checkout origin/master && yarn build` (you must have cloned from here)

flighter18, 2019-10-11:
@andi34 
I have done the following 

```
sudo -u www-data -s
git remote add andreknieriem https://github.com/andreknieriem/photobooth
git fetch andreknieriem
git checkout andreknieriem/master
yarn build
exit
```

but the buttons dosen't work.

andi34, 2019-10-11:
If no error was visible and it was telling `git checkout` was fine it should be ok. It'll tell you in terminal if it can not checkout because of uncommitted changes for example (in case you made some changes by hand in your source)

flighter18, 2019-10-11:
@andi34 no error was visible. so I think the button does still not work.

andi34, 2019-10-11:
You can check `git log` if latest commit hash matches the one here, should be `331d1ef4adeb4269893bb7bdc840fd0b2b1c6ec4` if you're not sure

flighter18, 2019-10-11:
@andi34 thank you for this hint.

```
[root@raspberrypi:/var/www/html# git log
commit 331d1ef4adeb4269893bb7bdc840fd0b2b1c6ec4 (HEAD, andreknieriem/master)
Author: sualko <klaus@jsxc.org>
Date:   Fri Oct 11 15:15:14 2019 +0200]
```
So it matches.

andi34, 2019-10-11:
You're able to test on Android?
Only specific browser?

flighter18, 2019-10-11:
i do not have any android device but i tested it with 3 browser on ipad and iphone.
none of them works.

andi34, 2019-10-11:
Ok. Might be able to check in the morning on Android. Will have to setup my pi again as access point for it.

andi34, 2019-10-11:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Quickly tested 😂 no issues on my phone , Android 9, Chrome browser, OnePlus 6.

[image]

flighter18, 2019-10-11:
I tested it now also on IOS 13. here it works too 🙈. So the problem is IOS 12

flighter18, 2019-10-11:
🥴 my ipad is to old for IOS13 😅 shit

andi34, 2019-10-11:
https://forum.webflow.com/t/button-not-responding-in-safari-on-iphone/51418
Seems a known issue?

commit 331d1ef4 (2019-10-11): fix gallery buttons on small screens

fix #116