andreknieriem/photobooth #117: Question on Performance

RaphaelKunis, 2019-10-10:
Hi, 
i am using photobooth for a while, first on a pi zero and now on a pi model 3B+.
If I use the plain git version it takes approximately between 8 and 11 seconds even on the 3B+ to show the picture in browser after the picture was taken by camera.

This is a bit long in my oppinion. 
How long does it take with your installation? 
Do i need to do some environment changes to my pi?

My current solution to solve this is 
- to totally disable gallery and image processing when gallery is not used (no thumbnail creation and no image changes), 
- to disable qr code creation and to turn of qr code creation when qr is disabled and 
- to set the picture quality to produce approximately 3 MB pictures. 

This leads to a processing time of 5 to 7 seconds.
My configuration is a Pi Model 3B+, with a Canon ES 600d camera.
Raspbian Buster
Gphoto: gphoto2 2.5.23, libgphoto2 2.5.23

sualko, 2019-10-11:
Here are some numbers, taken from a default installation (without collage/filter) on a Raspberry 3 (all times in seconds):

- Execution time `takePic.php`: 8.737024869918823
- Time until picture is taken and stored to disk: 1.5764682292938232
- Load image (imagecreatefromjpeg): 0.8934361934661865
- Resize for chromakeying: 4.136002063751221
- Thumbnail generation: 2.0890421867370605
- Write thumbnail: 0.01137089729309082

If I apply a filter the execution time rises to ~15.7 seconds which is way to much. Our aim should be to show the original image as fast as possible and to generate thumbnails and apply filters in the background. This will of course not go into 2.0, but I will consider it for 2.1. Thanks for reporting.

mhellmeier, 2019-10-13:
Jip, this question / issue is mostly based on the software part. I will test it with a new Raspberry Pi 4 Model B with 4GB RAM soon.

Remember that the SD card speed of your camera and raspberry pi is also in important point. With a fast r/w card you can get some speed improvements.

sualko, 2019-10-18:
I was able to make the image taking process a lot more responsive (preview is shown after 1.7s) and to improve the overall performance. As seen above one of the biggest issues was the resizing. Therefore I tried different solutions and as you can see, there are way faster methods (with a similar quality).

- GD (imagecopyresampled): 3.7123539447784
- GD (imagescale with triangle filter): 0.96465301513672
- Imagick (resizeImage with triangle filter): 1.6898829936981
- Convert: 3.9750638008118

With the new resize method, the thumbnail and chromakeying generation, are now ~75% faster:

- Resize for chromakeying: 1.1182799339294434
- Thumbnail generation: 0.48159098625183105

closed 2019-10-22 by sualko

mhellmeier, 2019-10-28:
> How long does it take with your installation?

I just tested it on a new Raspberry Pi Model 4B with 4GB of RAM and a pretty fast (and expensive 😆) SD card. It took around 9 seconds to take a normal picture and around 14 seconds to take a picture with a filter.

If I compare my results with the maximum of 11 seconds from @RaphaelKunis and 15.7 seconds from @sualko I would say that adjusting the hardware will also give a slightly performance improvement.

If you are interested I can also share my execution times after applying this PR.

_(Remember that I tested with version 2.0.2, that's before this PR was merged)._

sualko, 2019-10-30:
> If you are interested I can also share my execution times after applying this PR.

Would be nice to know. :+1:

PR #135 (2019-10-22): Optimize performance
- [x] separate trigger and post-process task
- [x] if possible use faster method to resize a picture
- [x] remove time measurement which was used for development

fix #117