andreknieriem/photobooth #120: disable gallery button when gallery stand alone is in use

danielovic295, 2019-10-11:
Hey guys,

first of all I have to tell you guys that I am really impressed and very thankful for what you are doing here. 

I think it would be great to have the possibility to disable the gallery button when I use the gallery stand alone, maybe via admin panel? 

Keep on the good work. 

Best regards,
Daniel

andi34, 2019-10-11:
More or less connected to https://github.com/andreknieriem/photobooth/pull/118

sualko, 2019-10-11:
@andi34 why are you always changing the title? I think the label is catchy enough 😂

@danielovic295 we will consider your request for one of the next releases and thanks for your kind words.

andi34, 2019-10-12:
@danielovic295 set `$config['show_gallery'] = false;` , gallery can be accessed via `localhost/gallery.php`

andi34, 2019-10-12:
https://github.com/andreknieriem/photobooth/blob/331d1ef4adeb4269893bb7bdc840fd0b2b1c6ec4/index.php#L42-L44

https://github.com/andreknieriem/photobooth/blob/331d1ef4adeb4269893bb7bdc840fd0b2b1c6ec4/index.php#L160-L162

Since this feature already exist we can close the issue/feature request.
Feel free to continue the discussion here.

closed 2019-10-12 by andi34