andreknieriem/photobooth #121: Installation instructions for Windows needed

andi34, 2019-10-11:
https://github.com/andreknieriem/photobooth/wiki/Installation-on-Windows

Maybe someone using Photobooth on Windows can help getting proper installation instructions?

Please leave a comment here if you're able to help :)

lagrandeours, 2020-01-19:
hello i try to make the install in a PC, but i'm more photographer than developper but i give a try and keep you inform

lagrandeours, 2020-02-21:
hello on windows i added some simple html code 
and it work fine to take the picture 
i added it in the core js bc in admin not working ...
$.post("http://localhost:5513/?slc=capture"

lagrandeours, 2020-02-21:
and now how to get the pictures in the workflow ? bc in digicam i can send the pictures to data/temp or img but nothing happend ...

andi34, 2020-02-21:
It should work out of the box and download the picture inside the tmp folder and doing the post processing to have the picture on the img folder as result. 

You only might need to adjust the take picture cmd via admin panel to match your setup:
`digicamcontrol\CameraControlCmd.exe /capture /filename %s`

Make sure the path matches your setup.
Maybe needs to be `..\digicamcontrol\CameraControlCmd.exe /capture /filename %s` ?

lagrandeours, 2020-02-22:
thank you so musch for your answear :-)
yes great with double dot :  ..\..\digiCamControl\CameraControlCmd.exe  /capture /filename %s 
and sorry to ask you about to use a Printer in windows what is the command ?

andi34, 2020-02-23:
Quick search gave 2 things you can try to print (scroll down on both pages an read the answers):
https://superuser.com/questions/1407593/how-to-print-using-cmd-only-with-the-name-of-the-printer

https://superuser.com/questions/1399939/how-to-print-image-file-in-windows7-using-command-line-lpr

andi34, 2020-02-23:
But `mspaint /p "%s"` should be enough.

```
Print an image from Paint directly through the command line by adding a space after "mspaint" and adding "/p" then a filename (e.g.: "mspaint /p c:/folder/subfolder/filename"). The command prints to your default printer. Use "/pt" followed by the name of a printer to print to a specific device.
```

lagrandeours, 2020-02-23:
ok thanks a lot may be %s was missing :-) 
but in fact it print only if i press the print button after taking a picture 
in the gallery the print button not working may be a little bug ? 

for the command to take the photo it s better to use Remote 
..\..\digiCamControl\CameraControlRemoteCmd.exe /c capture %s
with digicam running in background 
it take the photo quickly (with cmd it take time to connect to the camera for each picture) 

thanks a LOOOOOOOOOOOOOOOOOOOOOT 

this photobooth is my new project and with your app it s working just PERFECT !!! 

realy thanks how can i offer you a good Café :-)

andi34, 2020-02-23:
> ok thanks a lot may be %s was missing :-)
> but in fact it print only if i press the print button after taking a picture
> in the gallery the print button not working may be a little bug ?
> 

Anything visible in the console log of your browser why it fails?
(Press F12 to access it)

> for the command to take the photo it s better to use Remote
> ....\digiCamControl\CameraControlRemoteCmd.exe /c capture %s
> with digicam running in background
> it take the photo quickly (with cmd it take time to connect to the camera for each picture)
> 
> thanks a LOOOOOOOOOOOOOOOOOOOOOT
> 

Thanks a lot for the info :+1:  I'll try to setup on a Windows and play around myself for proper documentation of the installation instructions once I've more time.

> this photobooth is my new project and with your app it s working just PERFECT !!!
> 
> realy thanks how can i offer you a good Café :-)

No need, but if you like you can find a donation link in my fork https://github.com/andi34/photobooth or GitHub Userpage http://andi34.github.io/donations.html

lagrandeours, 2020-02-23:
http://localhost/photobooth-2.1.0/api/print.php?filename=images**%5C**20200223_182303.jpg

there is %5C in the code ....

andi34, 2020-02-24:
5C is the hex code for '\' (backward slash). I let you know if I've an idea how to solve it. Might be a windows specific issue.

andi34, 2020-02-24:
Does the Download link work?
Does Chromakeying link work?
QR from Gallery points to the right URL?

lagrandeours, 2020-02-24:
ok but when i send print command juste after taking the picture it work fine :-) 
this problem apear only from the gallery

lagrandeours, 2020-02-24:
juste see your reply i check this

andi34, 2020-02-24:
Just mobile and can't test. Maybe you have time and can give feedback. Please test https://github.com/andi34/photobooth/commit/04f94a8b20ccbdd3357ed0af661c10145caba778

lagrandeours, 2020-02-24:
Hi :)
so in dowload.php 
> echo file_get_contents(__DIR__.'/../data/images/'.$image);

Does the Download link work? 
no there is /../ and double /images folder ...

no <b>Warning</b>:  file_get_contents(C:\xampp2\htdocs\photobooth-2.1.0\api/../data/images/images\20200224_174845.jpg): failed to open stream: No such file or directory in <b>C:\xampp2\htdocs\photobooth-2.1.0\api\download.php</b> on line <b>7</b><br />

Does Chromakeying link work?
yes 
QR from Gallery points to the right URL?
no 
Does print work from Chromakeying?
yes it send to printer but seem to stuck after printing 
Tried a different browser? the same

andi34, 2020-02-24:
You've tried the code change posted above?

lagrandeours, 2020-02-24:
i m trying  this and the download work fine but ...

	echo file_get_contents(dirname(__DIR__,1).'/'.$image);
	// echo file_get_contents(__DIR__.'/../data/images/'.$image);

lagrandeours, 2020-02-24:
... qr code and print not ...

lagrandeours, 2020-02-24:
> You've tried the code change posted above?

yes same pb

lagrandeours, 2020-02-24:
may be in the print file this line 53 

  $url = 'http://'.$SERVER_IP.'/api/download.php?image=';
            QRcode::png($url.$filename, $filename_codes, QR_ECLEVEL_H, 10);

not good as my installation is in http://localhost/photobooth-2.1.0/

lagrandeours, 2020-02-24:
now it s ok the installation have to be in localhost direct

lagrandeours, 2020-02-24:
now i have to manage the printing issues with command line : print size / resize the pict etc ... 
i m trying with IrfanView 64 to manage différent settings ... 
I m so close :-)

andi34, 2020-02-24:
> now it s ok the installation have to be in localhost direct

Does it also work without my change now?
Path should be without `-` it `.`
There's an option "crop on print" In admin panel.

lagrandeours, 2020-02-24:
Does it also work without my change now?
NO not working ... keep the change :-)

Path should be without - it . ok ... ?

There's an option "crop on print" In admin panel. ok thanks

andi34, 2020-02-24:
> Path should be without - it . ok ... ?
> 

Can be a subfolder too like:
```photobooth
photobooth210```

Also I'd avoid whitespaces like
`photobooth 210`

lagrandeours, 2020-02-24:
juste a remarque about the QR Code from mobile it s better to open the picture directly in the browser and then download the picture manualy 
otherwise the download process not working directly from android i have to copy past the "text" in the browser or maybe it s due to the str_replace('\\','/',$filename) ... 

i have not test from iPhone ... 
(so i changed the qrcode.php) 
with 
$url = 'http://'.$SERVER_IP.'/';
QRcode::png($url.str_replace('\\','/',$filename), false, QR_ECLEVEL_H, 10);

andi34, 2020-02-24:
There must be an issue on your end with that, it's not getting the filename right. Double check if you've applied my changes correctly for the /\ replacement.

https://github.com/andreknieriem/photobooth/blob/master/api/qrcode.php

lagrandeours, 2020-02-24:
> > Path should be without - it . ok ... ?
> 
> Can be a subfolder too like:
> 
> ```
> photobooth
> photobooth210
> ```
> 
> Also I'd avoid whitespaces like
> `photobooth 210`

about the app in subfolder it's not working at all ...

andi34, 2020-02-24:
> > > Path should be without - it . ok ... ?
> > 
> > 
> > Can be a subfolder too like:
> > ```
> > photobooth
> > photobooth210
> > ```
> > 
> > 
> > Also I'd avoid whitespaces like
> > `photobooth 210`
> 
> about the app in subfolder it's not working at all ...

If it's inside a folder "photobooth" you'll have to access it via `localhost/photobooth`. I don't see a reason why this shouldn't work on windows.

lagrandeours, 2020-02-24:
do you have time to come on my pc with remote chrome access ?

andi34, 2020-02-24:
> do you have time to come on my pc with remote chrome access ?

Not today, maybe next days.

You can write a mail to andi@unlegacy-android.org and I'll give you a Photobooth zip you can test.

memberbro, 2020-08-20:
hello, i can't find the digiCamControl.zip file which i need to connect to the camera.
can someone send me a link or something like that?
thanks

andi34, 2020-08-20:
> hello, i can't find the digiCamControl.zip file which i need to connect to the camera.
> can someone send me a link or something like that?
> thanks

Mentioned inside the README ;)
http://digicamcontrol.com/

atilavilela, 2020-10-17:
I have this same problem @memberbro 
This is the error message that appears to me:
{error: "Gphoto returned with an error code", cmd: "digicamcontrol\CameraControlCmd.exe /capture /file…oth\data\tmp\1433bab09412343e0094a9fb551a8fc2.jpg", returnValue: 1, output: Array(0)}
cmd: "digicamcontrol\CameraControlCmd.exe /capture /filename C:\xampp\htdocs\photobooth\data\tmp\1433bab09412343e0094a9fb551a8fc2.jpg"
error: "Gphoto returned with an error code"

andi34, 2020-10-17:
Make sure the path exist

`C:\xampp\htdocs\photobooth\digicamcontrol\` with the `CameraControlCmd.exe` inside.

andi34, 2020-10-17:
Another thing might be that there's some other changes needed not included here atm.

https://github.com/andi34/photobooth/commit/629d8df9ec93a0447931745982a44743a6412153

https://github.com/andi34/photobooth/commit/233ca56272ee456ac2e7358ff658d6385ce3c908

https://github.com/andi34/photobooth/commit/83adf3fdb0b35991e7267bea35bfb57b3e088d81

https://github.com/andi34/photobooth/commit/6e66882c1f56d664633ace7c22ea6e5509aebafe

https://github.com/andi34/photobooth/commit/55651c20e707ddb31b465e1b57d0922d59a9b584

atilavilela, 2020-10-21:
fiz todo o processo e confirmei que o CameraControlCmd.exe está no local correto mas infelizmente não executou o arquivo. Tentei executar o comando /capture /filename %s direto no .exe mas também não tive sucesso

andi34, 2020-10-21:
`%s` is the path and filename, you'll have to type something like `/filename test.jpg` if you test it via console.

This issue is about creating instructions for Windows as there's no proper documentation. If you have success please let me know.

atilavilela, 2020-10-24:
@andi34 unsuccessful with the command manual. I intend to buy a raspbarry pi3 + the installation of the system in it is simpler than in windows?

andi34, 2020-10-25:
Your camera is supported by Digicamcontroll?
http://digicamcontrol.com/cameras

If not, a Pi only will make a difference if your camera is supported by gphoto2.
http://gphoto.org/proj/libgphoto2/support.php

atilavilela, 2020-10-25:
I have a canon sl2, when I run through the application it works normally but the commands do not work. can it be compatible with my computer? older motherboard and processor ....

andi34, 2020-10-26:
Actually does not fully work, it takes the picture, but failes at the applyEffects.php.
```
jquery.min.js:2 POST http://localhost/api/applyEffects.php 500 (Internal Server Error)
An error occurred error                      core.js:429 
```

Here's my setup:

1. Download needed files
Download Apache2 and Visual C++ Redistributable for Visual Studio from https://www.apachelounge.com/download/
Download PHP https://windows.php.net/download (VC15 x64 Thread Safe)
Download Digicamcontrol http://digicamcontrol.com/download
Download Notepad++ https://notepad-plus-plus.org/downloads
Download latest Photobooth release https://github.com/andi34/photobooth/releases

2. Install needed Software
Install Notepad++
Install Visual C++ Redistributable for Visual Studio
Extract the Apache2 ZIP to C:/
Extract the PHP ZIP to C:/php7

3. Prepare Apache2
Edit `C:\Apache24\conf\httpd.conf using Notepad++`

Find the following text (~ on line 284):
```
<IfModule dir_module>
    DirectoryIndex index.html
</IfModule>
```
And change it to  
```
<IfModule dir_module>
    DirectoryIndex index.html index.php
</IfModule>
```

To the bottom of the file add the following:  
```
LoadModule php7_module "C:/php7/php7apache2_4.dll"
<FilesMatch \.php$>
    SetHandler application/x-httpd-php
</FilesMatch>

PHPIniDir "C:/php7"
```

Inside `C:\Apache24\htdocs` add a new file called `info.php`  and add the following content:  
```
<?php
phpinfo();
?>
```
Inside C:\Apache24 create a new file called `cmd.bat` and add th following content:
```
cd "C:\Apache24\bin"
cmd
```

4. Prepare PHP
Go to `C:\php7` and create a copy of the `php.ini-production` and rename it to `php.ini`

Edit the `php.ini` using Notepad++:  
Find `;extension=gd2` and remove the `;` in front of the line.

Find `;extension_dir = "ext"` and remove the `;` in front of the line.

5. Start Apache Server
Go to `C:\Apache24` and right click on the cmd.bat, choose "Run as administrator".  
First time, type `httpd.exe -k install`

Once that's done, lets start our webserver:  
`httpd.exe -k start`

6. Test your Webserver & PHP

Open [http://localhost/info.php](http://localhost/info.php) in your Browser, you should see the PHP Information page.

7. Install Digicamcontrol

Install to `C:\Apache24\htdocs\digicamcontrol\`

8. Setup Photobooth
Remove all files inside `C:\Apache24\htdocs\`
Extract the Photobooth ZIP to `C:\Apache24\htdocs\`

Open [http://localhost/adminp](http://localhost/admin) in your Browser.

Adjust your take picture command:  
`C:\Apache24\htdocs\digicamcontrol\CameraControlCmd.exe /capture /filename %s`

andi34, 2020-10-26:
Not sure what I am missing or what could be wrong on the applyEffects.php...

Anyone an idea?

andi34, 2020-10-27:
Even after the PC was turned off it wasn't working.
I've stopped the Apache Server via `httpd.exe -k stop`, waited a minute and started it again `httpd.exe -k start` and its working now. Crazy.

andi34, 2020-10-27:
@atilavilela I've updated the first post with my instructions. Could you please test if it also works for you?

closed 2022-10-04 by andi34