andreknieriem/photobooth #122: Neue Sprache (Griechisch)

fossil74, 2019-10-12:
Hallo Leute,
weiß nicht ob es hier an der Richtigen stelle gepostet ist. Darum löschen wenn Falsch.

Hier eine neue Sprache für das Photobooth.
[Griechisch](https://github.com/fossil74/Photobooth/blob/master/resources/lang/gr.js)

andreknieriem, 2019-10-12:
Mega! Vielen Dank!

andi34, 2019-10-12:
@fossil74 vielen Dank! Ich habe entsprechend eine passende Pullrequest erstellt.

> Mein Griechisch ist nicht Perfekt daher mit einem Online Übersetzer gemacht. Bitte ändern wenn was falsch ist.
> 

Ist dein griechisch so gut, das du bestätigen kannst das alles richtig ist? Ein gewisser Qualitätsstandard sollte natürlich gegeben sein wenn neue sprachen hinzukommen.

fossil74, 2019-10-12:
@andi34 
bin selbst Grieche aber in Deutschland geboren. Also ich kann bestätigen das es zu 90% richtig ist. 
Manches kann man nicht so Übersetzen wie es da steht, Galerie zb. gibts in GR nicht in dieser Form wie in DE oder in US. 

Aber ich lass gerne nochmal einen zweiten drüber schauen wenn Ihr wollt, ist kein problem das zu machen.

andi34, 2019-10-12:
@fossil74 alles gut, wollte nur nochmal nachhaken

andi34, 2019-10-12:
Achso, muss auch nicht 1:1 sein, glaube statt Gallery wäre `Συλλογή` passend, oder?

flighter18, 2019-10-12:
ich würde sonst noch eine Griechen kennen, die in Griechenland lebt.
Villeicht könnte sie nochmals darüber schauen?

andi34, 2019-10-12:
Vllt kann man in Zukunft auf crowdin zurückgreifen für Übersetzungen. Im Zweifelsfall müsste man schauen ob das format angepasst werden kann.

https://crowdin.com/page/open-source-project-setup-request

Habe crowdin für 2 große Android Projekte gemanaged in der Vergangenheit, müsste mich dann nur wieder reinfuchsen.

Generell gilt: man kann nicht alles 1:1 übersetzen und muss man auch nicht. So wie es im entsprechenden Land im alltäglichen Sprachgebrauch benutzt wird.
Soweit ich weiß würde man es aus dem griechischen ins deutsche übersetzen würde man "Sammlung" anstelle von "Gallery" nehmen.

fossil74, 2019-10-12:
> Achso, muss auch nicht 1:1 sein, glaube statt Gallery wäre `Συλλογή` passend, oder?

Wenn dann so Φωτογραφική συλλογή

sualko, 2019-10-14:
@fossil74 thanks for your contribution :tada:. If you have further chances, just open the [gr.js](https://github.com/andreknieriem/photobooth/blob/master/resources/lang/gr.js) file and click on the pencil to create a new pull request. 

[image]

PR #123 (2019-10-14): Add greek language
close #122
closed 2019-10-14 by sualko