andreknieriem/photobooth #125: Fallback to English if a translation is missing

andi34, 2019-10-14:


andi34, 2019-10-21:
@sualko if we add everywhere above (or below?) to the config related lines
`<script type="text/javascript" src="resources/lang/en.js"></script>`

So it'll look like
```
<script type="text/javascript" src="resources/lang/en.js"></script>
<script type="text/javascript" src="resources/lang/<?php echo $config['language']; ?>.js"></script>
```
Is smart enough? You've a better idea to solve it?

rawbertp, 2019-11-02:
I wouldn't try to reinvent the wheel but use an existing I18N framework instead. This might be a good read: https://medium.com/best-of-i18n/the-best-javascript-i18n-libraries-657f9fd2124a

andi34, 2019-11-03:
@rawbertp you're working on it already?

rawbertp, 2019-11-03:
@andi34 Not really. I spent some thoughts on it, though. Let me share these thoughts ;)

* First of all, I do not know the reasons for offering that many translations. It might be an option to just offer English and German and get rid of any other translations which are not fully backed by at least one "maintainer".
* Providing translations for EN and DE would be mandatory. Any other translations "best effort" with the corresponding maintainers being responsible.
* As for the fallback: Taking the above mentioned into consideration, I think there is actually no real need for falling back to EN. EN and DE must be 100% complete; if translations are missing for Greek, @dimitrixyz will be responsible for providing the missing translations. Or anyone else, being able to contribute. A bad "Google Translate" translation is no better than displaying the key (IMHO again).
* With the major redesign going on, it might be worth, implementing an OSS i18n library but for now I think this has no real priority.
* And again: I suggest getting rid of any languages the main contributors here are not able to provide translations for.

andi34, 2019-11-03:
That's why I like to have a fallback to English, it would make the missing translations visible for contributors using Photobooth in their mothers language. 
If there's no updated translations for a long time we can remove them until they are up to date again.

andi34, 2019-11-03:
My thought of calling the en.js after the language.js does not work btw.
Need to find a smarter solution

andi34, 2019-11-03:
https://github.com/andreknieriem/photobooth/blob/master/resources/js/l10n.js#L11

How to make it check if the key exist in en.js and use it if it exists?

rawbertp, 2019-11-03:
No a JS expert but it appears you can't load another JS like that.

Maybe change the `<lang>.js` to `json` files and programmatically load them. It should be easy then to implement a fallback.

andi34, 2019-11-03:
Same here, not an expert on that...
and yeah... can't get it working to get the translation from en.js instead. Maybe @sualko has an idea how to get this solved.

rawbertp, 2019-11-03:
@andi34  See #179 for a very easy (potential) solution to this problem.

andi34, 2019-12-18:
https://github.com/philipptrenz/photo-booth/pull/95/commits/b37cebf1e8178adf366f67e0581d5f15d65924cc

Not sure when I find time, but looks like a solution which we might be able to use too

andi34, 2020-01-16:
Let's delay this to v2.1.1, 
v2.1.0 should be good to go now

andi34, 2022-10-04:
Fixed since Photobooth v3
https://photoboothproject.github.io/Changelog

closed 2022-10-04 by andi34