andreknieriem/photobooth #128: Probleme mit dem Hardware Button

F-Major, 2019-10-15:
Ich habe massive Probleme einen Hardware Button zu integrieren. 
Ich habe bereits in der Admin Oberfläche '13' eingetragen für KEY_ENTER sodass das mit dem button.py übereinstimmt. Nun habe ich auch schon den GPIO 24 ausgelesen welcher mit dem Taster auch 0 oder 1 richtig anzeigt. 
Die Probleme entstehen mit uinput und das er kein device findet.

Ich würde mich freuen wenn  mir jemand weiterhelfen könnte.

sualko, 2019-10-15:
> Ich habe massive Probleme einen Hardware Button zu integrieren.

Please post any error messages you get.

> Ich habe bereits in der Admin Oberfläche '13' eingetragen

Does it work if you press enter on a normal keyboard?

> Die Probleme entstehen mit uinput und das er kein device findet.

Did you load the uinput kernel module via `modprobe -i uinput`?

F-Major, 2019-10-15:
`pi@raspberrypi:~ $ modprobe -i uinput
modprobe: ERROR: could not insert 'uinput': Operation not permitted
pi@raspberrypi:~ $ python /var/www/html/button.py
Button Pressed
Button Pressed
Traceback (most recent call last):
  File "/var/www/html/button.py", line 17, in <module>
    with uinput.Device([uinput.KEY_ENTER]) as device:
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 178, in __init__
    self.__uinput_fd = fd or fdopen()
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 84, in fdopen
    return _libsuinput.suinput_open()
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 70, in _open_error_handler
    raise OSError(code, msg)
OSError: [Errno 19] Failed to open the uinput device: No such device
pi@raspberrypi:~ $
`

F-Major, 2019-10-15:
> Does it work if you press enter on a normal keyboard?

ja wenn ich die ENTER taste auf der tastatur drücke funktioniert es

F-Major, 2019-10-15:
`#!/usr/bin/env python
# Sends alt+p via GPIO24 to take a new picture
import RPi.GPIO as GPIO
import time
import os
import uinput

print('Button Pressed')

GPIO.setmode(GPIO.BCM)
GPIO.setup(24, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    input_state = GPIO.input(24)
    if input_state == False:
        print('Button Pressed') #<- for debugging only
        with uinput.Device([uinput.KEY_ENTER]) as device:
         time.sleep(1)
         device.emit_combo([uinput.KEY_ENTER])
         time.sleep(10)

`
Das ist der button.py

`# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command

@reboot python /var/www/html/button.py &

`
und das ist dort drin.

andi34, 2019-10-15:
You've tried with `sudo` in front?

Not sure if still actually needed to add uinput to your modules file meaning that it will be loaded every boot.
```
sudo sh -c "echo uinput >> /etc/modules"
```

F-Major, 2019-10-15:
Ich habe es nochmal mit sudo ausprobiert und leider keinen Erfolg. 

> Not sure if still actually needed to add uinput to your modules file meaning that it will be loaded every boot.

sudo sh -c "echo uinput >> /etc/modules"

Habe ich auch ausprobiert funktioniert auch nicht.

sualko, 2019-10-15:
try `sudo modprobe -i uinput`. If it still doesn't work try 

```
sudo apt install libudev-dev
sudo pip install python-uinput
```

F-Major, 2019-10-16:
`pi@raspberrypi:~ $ sudo apt install libudev-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libudev-dev is already the newest version (232-25+deb9u12).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ sudo pip install python-uinput
Requirement already satisfied: python-uinput in /usr/local/lib/python2.7/dist-packages
pi@raspberrypi:~ $ sudo modprobe -i uinput
pi@raspberrypi:~ $ python /var/www/html/button.py
Button Pressed
Button Pressed
Traceback (most recent call last):
  File "/var/www/html/button.py", line 17, in <module>
    with uinput.Device([uinput.KEY_ENTER]) as device:
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 178, in __init__
    self.__uinput_fd = fd or fdopen()
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 84, in fdopen
    return _libsuinput.suinput_open()
  File "/usr/local/lib/python2.7/dist-packages/uinput/__init__.py", line 70, in _open_error_handler
    raise OSError(code, msg)
OSError: [Errno 13] Failed to open the uinput device: Permission denied
pi@raspberrypi:~ $

`
Immer noch keine Funktion..

sualko, 2019-10-18:
I was able to reproduce the issue and now updated the wiki with the [correct instructions](https://github.com/andreknieriem/photobooth/wiki/FAQ#can-i-use-hardware-button-to-take-a-picture-on-my-raspberry-pi). 

FYI the `No such device` message means the kernel module is not loaded and `Permission denied` means you didn't call the script as root.

closed 2019-10-18 by sualko

F-Major, 2019-10-18:
Vielen Dank. 
ich bekomme nach `pi@raspberrypi:~ $ lsmod | grep uinput
uinput                 20480  1
`
was raus aber trotzdem funktioniert er immer noch nicht als Auslöser für die photobooth

PeterHaase1995, 2020-12-20:
Hallo zusammen,
Ich habe leider das selbe Problem. Ich bekomme über pi@raspberrypi:/var/www/html $ sudo python button.py  zwar immer das Ergebnis "Button Presses" es passiert sonst aber nichts.. Das Auslösen über die Enter Taste auf der Tastatur funktioniert problemlos.
Vielen Dank für eure Hilfe!

andi34, 2020-12-20:
Rufst du die Photobooth am Pi selbst auf? (Muss über `localhost` oder `127.0.0.1` aufgerufen werden.)

jacques42, 2020-12-20:
Hi - Versuch es bitte mal mit einer Installation aus dem aktuelleren Fork von andi34/photobooth. Falls das Problem noch existiert, mach dort bitte ein Issue auf und ich schau mir das gerne an

Grüße

PeterHaase1995, 2020-12-20:
vielen dank für die schnellen antworten!
ja, ich rufe die Photobooth über den pi auf. Ich vermute, dass das uinput Modul nicht richtig arbeitet. Das "button.py" Skript läuft, es gibt aber kein "Enter" aus.
Die Installation habe ich step by step nach Anleitung gemacht.

PeterHaase1995, 2020-12-20:
ok, das war wohl das Problem... jetzt funktioniert es
vielen Dank für eure Hilfe!