andreknieriem/photobooth #129: CheeseTime implemented "incorrectly"

Norman-Sch, 2019-10-16:
Take.Pic should be run immediately after the countdown. In version 2.0, however, Take.Pic is triggered after CheeseTime expires. This leads to unnecessary delays because until the camera has focused, a certain amount of time passes. When the camera is triggered, "Cheeese" should be displayed. With CheeseTime, this time can be customized.

Hopefully there will be a quick fix, because Saturday is a wedding again :)

andi34, 2019-10-16:
Please use the template to open an issue for future issues.

You could try changing the delay here from 500 to something smaller:
https://github.com/andreknieriem/photobooth/blob/18a61df550a3319efe23bbcae16d863ef0d4676d/resources/js/core.js#L171-L174

Or remove the following lines in resources/js/core.js:
https://github.com/andreknieriem/photobooth/blob/18a61df550a3319efe23bbcae16d863ef0d4676d/resources/js/core.js#L171
https://github.com/andreknieriem/photobooth/blob/18a61df550a3319efe23bbcae16d863ef0d4676d/resources/js/core.js#L174
https://github.com/andreknieriem/photobooth/blob/18a61df550a3319efe23bbcae16d863ef0d4676d/resources/js/core.js#L187

Let me know if it solves your problem.

Norman-Sch, 2019-10-16:
> Another idea:
> We could try calling the cheese from takePic function instead takePic from cheese function.

Thank you for the tip. For me this solves the problem, but its not that nice. maybe somebody could do better...I am no programmer :D

andi34, 2019-10-16:
@Norman-Sch could you please test https://github.com/andreknieriem/photobooth/pull/133 ?
```
git fetch origin
git checkout origin/issue129
```

sualko, 2019-10-16:
I would not say "incorrectly", but I understand what you mean. If we follow your proposal and someone sets `cheese_time` to 5 seconds, but the picture was taken after 2 the cheeeese message would be still there or we would violente the config. The issue is that we can't predict when a picture was taken. The assumption is that the camera triggers after half a second (`processingDelay`) and the `cheese_time` is just the foreplay. See also #117 to understand the problem in more detail.

My recommendation is that you set `cheese_time` to 500ms and adjust [core.js#L174](https://github.com/andreknieriem/photobooth/blob/34eb504ccb6c864e3dee060da49d04515f2dd4e6/resources/js/core.js#L174) to a good approximation of the time your camera needs to trigger the shutter. Maybe someone makes a pr to adjust this value via the admin settings :wink:, or you have to wait until I did some performance optimizations. The idea is to separate the picture task from the post-processing. This way the takePic request would immediately resolve after a picture was taken.

Btw. if you use this project to make some money it would be nice if you could give something back to the people. One possibility would be a donation to [my local red cross community](https://drk-tettnang.de/aktiv-werden/spenden.html), but don't feel pressured. This is open source :tada:

andi34, 2019-10-17:
> Btw. if you use this project to make some money it would be nice if you could give something back to the people. One possibility would be a donation to [my local red cross community](https://drk-tettnang.de/aktiv-werden/spenden.html), but don't feel pressured. This is open source

👍 That's a good thing! That's something I told my Android users too (http://andi34.github.io/donations.html).

andi34, 2019-11-05:
Performance optimizations merged. Can be reopened if there's still an issue with it.

closed 2019-11-05 by andi34