andreknieriem/photobooth #130: Problem mit Hochzeitssymbol

obkram, 2019-10-16:
Ich habe seit gestern ein Problem welches ich heute bei einem frisch aufgesetzten System immer noch nachvollziehen kann.
Ich kann das Symbol für Hochzeiten nur auswählen und dauerhaft speichern wenn keine Namen eingetragen sind. Sobald ich einen Namen eintrage und speichere steht dort wieder Symbol wählen. Einzige Änderung zum Standart ist das Livewallpaper. Kann das jemand bestätigen

andi34, 2019-10-16:
Please use the template and fill it instead removing everything:
```
**Describe the bug**
<!-- A clear and concise description of what the bug is. -->

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
<!-- A clear and concise description of what you expected to happen. -->

**Screenshots**
<!-- If applicable, add screenshots to help explain your problem. -->

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Photobooth**
- Did it work before? [yes/no]
- Which version of Photobooth are you using?

**Additional context**
<!-- Add any other context about the problem here. -->
```

andi34, 2019-10-16:
How does your `config/my.config.inc.php` look like? Please remove mail parts if added to not share them public.

andi34, 2019-10-16:
issue seems to be mentioned below the URL

[image]

obkram, 2019-10-16:
Also meine my.config.inc.php sieht wie folgt:
<?php
$config = array (
  'is_wedding' => true,
  'wedding' => 
  array (
    'groom' => 'S',
    'bride' => 'A',
  ),
);

obkram, 2019-10-16:
Sorry, der Rest wurde irgendwie geschluckt. hoffe es klappt jetzt.
# <?php
# $config = array (
# 'is_wedding' => true,
# 'wedding' => 
#  array (
#   'groom' => 'S',
#    'bride' => 'A',
#  ),
#);

andi34, 2019-10-16:
@obkram https://github.com/andreknieriem/photobooth/commit/74c6c2cb116240abd0fe120c87483ba1d06c6470 is not a proper fix.

This could happen on other options too, but it should work for now here.
We need to save the whole array if one part of an array is changed, atm it excludes the default config from the array,

Here it must look like this:
```
  'wedding' => 
  array (
    'groom' => 'Name 2',
    'bride' => 'Name 1',
    'symbol' => 'fa-heart-o',
  ),
```

PR #134 (2019-10-18): merge default and custom config deeply
fix #130
obsoletes #131
commit a733d8f1 (2019-10-18): merge default and custom config deeply

fix #130
obsoletes #131
closed 2019-10-18 by sualko