andreknieriem/photobooth #136: Brauche kleinen Anstoß

mdkhome, 2019-10-17:
Hi, ich kenne mich nicht gut in der Sache aus, aber ich versuche mein bestes 💃 
Ich glaube ich habe irgendwo einen Gedankenfehler, komme aber nicht drauf. Währe super wenn einer mir helfen könnte.

Denke das ich nach der Anleitung gegangen bin, aber egal wie oft ich mein Rasperry neu aufsetze, komme ich nicht weiter.

Immer wenn ich den Respi über browser aufrufe, bekomme ich keine richtige Grafikseite angezeigt sondern nur links. Und unter /Admin kann ich nix ändern da ich kein speichern Button oder so was finde.
 Ich bedenke mich schon mal für die Hilfe.
[image]

sualko, 2019-10-17:
Please use the issue template and provide the log of all steps you performed.

closed 2019-10-17 by sualko

mdkhome, 2019-10-17:
Also ich weiß nicht weiter, ich habe mich an die Anleitung gehalten. Immer und immer wieder. Aber immer das selbe Ergebnis, Keine Idee woran es liegen könnte? Nutze Respberry pi 3b+ mit Strech Desktop. Habe alle Updates und Upgrades durchgeführt....

sualko, 2019-10-17:
> Immer und immer wieder

I believe you, but I have no crystal ball or superpowers (I have, but I'm not allowed to talk about it :wink:). Without logs we can't debug this.

> Keine Idee woran es liegen könnte?

You didn't create the css files, but without logs I can't tell you why this failed.

andi34, 2019-10-17:
> Also ich weiß nicht weiter, ich habe mich an die Anleitung gehalten. Immer und immer wieder. Aber immer das selbe Ergebnis, Keine Idee woran es liegen könnte? Nutze Respberry pi 3b+ mit Strech Desktop. Habe alle Updates und Upgrades durchgeführt....

Yarn installed?
Any errors while running `yarn install` or `yarn build`?

mdkhome, 2019-10-17:
OK, dachte du würdest deine Glaskugel nicht nur für Reichtum und Frauen nutzen ;)
Welches Log brauchst du/ihr denn? Bin nicht sehr bewandert in sachen Linux.
Muss ich noch was erstellen, dachte hier wäre alles startbereit.?!

andi34, 2019-10-17:
> OK, dachte du würdest deine Glaskugel nicht nur für Reichtum und Frauen nutzen ;)
> Welches Log brauchst du/ihr denn? Bin nicht sehr bewandert in sachen Linux.
> Muss ich noch was erstellen, dachte hier wäre alles startbereit.?!

https://github.com/andreknieriem/photobooth/wiki/Installation-on-Debian

mdkhome, 2019-10-17:
OK danke dafür schonmal, ich versuche es mit dem Tutorial mal

andi34, 2019-10-17:
> OK danke dafür schonmal, ich versuche es mit dem Tutorial mal

It's mentioned in the README to follow the instructions. Also updating from 1.9 or older should be done by a clean new setup as there have been big changes in all parts of the source, also security improvements (no need for sudo rights anymore to take pictures).

mdkhome, 2019-10-17:
Stretch hatte ich nur Installiert da die meisten Tutorials die ich so gefunden hatte auf Stretch aufbauen. Aber mit dem Link von Andi34 versuch ich nun erneut mein Glück. Schon mal danke an alle die mir Rückmeldung gegeben hatten.

@obkram: Die IP Adresse ist mein lokales Netzwerk, worüber auch der PI angeschlossen ist

andi34, 2019-10-17:
@mdkhome am besten immer die hier in der Repo angegeben Instruktionen folgen, wir halten das hier immer aktuell damit es zum Source passt.

mdkhome, 2019-10-17:
Soeben hab ich es aufgespielt, läuft alles wie am Schnürchen, vielen Dank.