andreknieriem/photobooth #138: Alte IOS Version

F-Major, 2019-10-18:
**Describe the bug**
Mein IPad 3 mit der IOS Version 9.3.5 kann Photobooth nicht im Browser (Safari) öffnen 

**Expected behavior**

**Screenshots**
https://user-images.githubusercontent.com/35504605/67089458-eb3d3400-f1a7-11e9-854d-707a865287cb.jpg

**Environment (please complete the following information):**
 - OS: iOS 9.3.5
 - Browser safari
 - Version 

**Photobooth**
- Did it work before? yes
- Which version of Photobooth are you using? 2.0.0

**Additional context**

sualko, 2019-10-18:
I don't own an iOS device, but do you have the possibility to open a debug console? Because we need some more information. I guess the browser is too old, so it would be great if you could also provide the version of safari.

andi34, 2019-10-18:
@sualko https://github.com/andreknieriem/photobooth/issues/116
iOS 12 also still has issues with buttons. Maybe it's same source of problem here?

F-Major, 2019-10-18:
https://user-images.githubusercontent.com/35504605/67105574-492e4380-f1c9-11e9-856b-1548efb9ebcd.jpg

Ist die Versoin einfach zu alt?

andi34, 2019-10-18:
Tried updating Safari?

sualko, 2019-10-18:
> Ist die Versoin einfach zu alt?

Yes that's too old. As an open source developer we don't have the resources to support every browser version.

closed 2019-10-18 by sualko