andreknieriem/photobooth #139: Kleinigkeiten

markussiegert, 2019-10-18:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

**Screenshots**

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Photobooth**
- Did it work before? [yes/no]
- Which version of Photobooth are you using?

**Additional context**

sualko, 2019-10-18:
> ich kann die Weboberfläche in Googles Chrome nicht darstellen, ist für mich aber nicht so das Problem

Please open a new issue for this and provide a proper javascript log from your browser (use ctrl+I the debug console).

> mir ist aufgefallen das, wenn man die Bilder über die Adminseite löscht, die links in der Galerie erhalten bleiben.

I can confirm this issue and will provide a fix asap.

> da ich gern das Hintergrundbild bei Bedarf ändere, lässt sich das aufgrund fehlender Rechte als User pi nicht mehr realisieren.

An easy fix would be the following, assuming you installed photobooth inside `/var/www/html`.

```
sudo chgrp -R pi /var/www/html/resources/
sudo chmod -R g+w /var/www/html/resources/
```

closed 2019-10-18 by sualko

markussiegert, 2019-10-18:
Thanks a lot for your work and help.

obkram, 2019-10-18:
#mir ist aufgefallen das, wenn man die Bilder über die Adminseite löscht, die links in der Galerie erhalten bleiben.

Das kann ich bestätigen. Ist mir auch aufgefallen.

# da ich gern das Hintergrundbild bei Bedarf ändere, lässt sich das aufgrund fehlender Rechte als User pi nicht mehr realisieren.
Ich weis nicht ob es sich hierbei um Fehler handelt oder ich einfach keine Ahnung habe wie es einfach geht.

Wenn du den PhotoBooth auf einem RPI laufen hast dann öffne die Console und gib...
sudo pcmanfm
... ein. Dann wird der Filemanager mit Adminrechten geöffnet und du kannst die Files ändern und bearbeiten

commit 54fd4eb1 (2019-10-18): fix deletion of db file

fix #139