andreknieriem/photobooth #141: Keine Darstellung in Google Chrome

markussiegert, 2019-10-18:
ich kann die Weboberfläche nicht im Chrome anzeigen lassen.
in Edge und Firefox geht es.
Wurde gebeten ein log mitzugeben.

Photbooth Version 2

[log.txt](https://github.com/andreknieriem/photobooth/files/3744021/log.txt)

[image]

sualko, 2019-10-18:
Did you make any modifications? Because it seams that you load some javascript files twice.

Can you go to the network tab, inside the developer console, right click on any item, choose "Save all as HAR with content" and upload the file here, or send it to klaus@jsxc.org?

markussiegert, 2019-10-18:
First of all thanks for the quick help.
No, I think I installed everything according to the instructions and in
other browsers it works.
The desired file is in the appendix.
I wish you a nice evening and a nice weekend.
Yours sincerely
Markus Siegert

sualko, 2019-10-18:
@markussiegert if you reply to the github notification the attachment gets removed. Please send it directly to klaus@jsxc.org, or upload it directly to this issue.

markussiegert, 2019-10-18:
if send it.

andi34, 2019-10-18:
Tested on Chrome 2 days ago:
> **Screenshots**
> [image]
> 
> **Environment (please complete the following information):**
> 
> * OS: Xubuntu
> * Browser : Chrome
> * Version:  77.0.3865.90 (Official Build) (64-bit)

sualko, 2019-10-18:
The log looks good. Which version of chrome do you use? Can you try to [start chrome without extensions](https://www.techsupportall.com/start-google-chrome-without-extensions/)?

markussiegert, 2019-10-18:
I tried it in incognito mode and it works. Thanks for your help.

closed 2019-10-18 by andi34