andreknieriem/photobooth #144: No saving after chroma keying

obkram, 2019-10-18:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

**Screenshots**

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Photobooth**
- Did it work before? [yes/no]
- Which version of Photobooth are you using?

**Additional context**

obkram, 2019-10-18:
Sorry, I pressed the enterkey to early .
Take a picture
Go to gallery
Select the picture
Select chroma keying
Select a Background and press save

Error: there will always shown save an nothing more will happen.

Aktuelles Raspbian
Aktueller Standart-Browser von Raspbian

sualko, 2019-10-18:
Thanks for your report. There was an include missing.

sualko, 2019-10-18:
Btw, do you know why the preview is that small? It could be way bigger.

closed 2019-10-18 by sualko

obkram, 2019-10-18:
@sualko you are very quickly. I tested it on my system. The changes will work fine. Thank you.

commit b66ad7d2 (2019-10-18): add missing include

fix #144