andreknieriem/photobooth #146: Enhance chroma keying

sualko, 2019-10-18:
I did some tests with an example image which I found in the internet and I think the result is very disappointing. A quick test with [seriously.js](https://github.com/brianchirls/Seriously.js/) looked much better. The library was not updated in the last years, but the current library is even not that active :wink:. I think we could switch libraries, if we look at #126. One think we should test before, if seriously.js is also running on a raspberry, because I think it is using webgl.

Original image:
[image]

Current implementation:
[image]

Seriously.js:
[image]

andi34, 2019-10-19:
Wow, that's a big difference.

andi34, 2022-10-04:
Available since Photobooth v3
https://photoboothproject.github.io/Changelog

closed 2022-10-04 by andi34