andreknieriem/photobooth #148: Reloading Livewallpaper

obkram, 2019-10-19:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

obkram, 2019-10-19:
Mein Text wurde leider nicht übernommen, also nochmal...
In der Mainpage des Photo Booth wird das Livewallpaper nicht jedes Mal geladen. Bei der Adminpage jedoch immer.

Gibt es die Möglichkeit ein Neuladen des Background in der Mainpage des Photobooth zu erzwingen, denn nach einer Rückkehr von der Gallery oder der Adminpage zur Mainpage ist das Livewallpaper meist weg und kommt erst nach dem 6ten oder 7ten Aufruf der Seite wieder

Vielleicht könntet ihr auf der Adminpage im Bereich Ordner noch die Möglichkeit schaffen die Quelle des Wallpapers bzw. der URL anzugeben.

Ich denke das das Livewallpaper kaum von jemandem genutzt wird, also hat die Anfrage hier weder Priorität noch muss sich jemand verpflichtet fühlen hier eine Lösung zu schaffe. Ich danke euch trotzdem im Voraus

obkram, 2019-10-19:
/Users/sven/Desktop/Image-1.jpg

andi34, 2019-10-19:
Edit function ;)

[image]

andi34, 2019-10-19:
du hast die `` stehen lassen, hierdurch wurde dein text ausgeblendet. habe das fuer dich gefixt.

obkram, 2019-10-19:
Vielen Dank. Zu meiner Schande muss ich gestehen das mir eben gemeldet wurde das auf meinen Mac der Safari für Github zu alt ist. Edits und Bilderuploads funktionierten deshalb leider nicht. Hab auf Firefox gewechselt. Sieht besser aus.

[image]

andi34, 2019-10-19:
Kannst du den Stream neu starten unabhängig von der Booth?

obkram, 2019-10-19:
Also ich habe den Stream auf einem 2ten Browserfenster zum Testen permanent anliegen gehabt. Das heisst der Stream liegt an der Netzwerkschnittstelle meines Rechners devinitiv an und musste nicht noch angefordert werden.
Ich gehe ja davon aus das das Endgerät den Stream anfordert und nicht der RPI mit Photobooth, da ich unter URL in der style.css ja die Zeiladresse des Motionservers eintragen muss. Ist in meinem Fall auch der Rpi aber Localhost funktioniert ja nur vom Aufruf des Browsers vom RPI aus.
Auch ein Neustart der Streams in Motioneye bringt keine Änderung des Wallpapers

closed 2020-10-20 by obkram