andreknieriem/photobooth #150: Two suggestions for Chromakeying

obkram, 2019-10-19:
Leider ist bei Englisch zu bescheiden um es zu beschreiben:

1. Nach der Aufnahme eines Bildes wird man auf die Seite des gerade aufgenommenen Bildes weitergeleitet. Von dort aus muss man erst in die Galerie wechseln um Chromakeying ausführen zu können. Mein Vorschlag wäre den Gallerybutton unten in der Viererdarstellung durch die Weiterleitung ( Pinsel ) nach Chromakeying zu ersetzen.

2. Befindet man sich in der Gallery und man hat zuvor ein Bild mit Chromakeying gespeichert dann ist es möglich dieses erneut in der Gallery auszuwählen. Ist dieses bereits bearbeitete Bild dann geöffnet und man versucht erneut Chromakeying anzuwählen dann erscheint eine Fehlermeldung. Vielleicht könnte man die erneute Nachbearbeitung eines mit Chromakeying bearbeiteten Bildes in der Gallery unterdrücken, es in der Gallery als bearbeitet kennzeichnen ( Rahmenfarbe ) oder zumindest den Text der Fehlermeldung ändern in .. 
Eine nochmalige Bearbeitung ist leider nicht möglich

Danke ihr Lieben

andi34, 2019-10-19:
Duplicate https://github.com/andreknieriem/photobooth/issues/126

closed 2019-10-19 by andi34

obkram, 2019-10-19:
Punkt 1. sehe ich ein. Da hast du recht, das gibt es wohl schon
wie sieht es denn mit Punkt 2. aus? Unterbindung einer erneuten Bearbeitung mit Chromakeying von bereits mit Chromakeying bearbeiteten Bildern?
[image]

andi34, 2019-10-19:
Füg das gerne da noch mit ein. Geht ja darum, dass der komplette Workflow überarbeitet werden soll.

andi34, 2019-10-19:
Bezüglich des Workflow überarbeiten:
Direkt nach Erstellung des Bildes wird der Hintergrund gewählt.

 Inwieweit von der Gallerie aus dann noch die Möglichkeit für keying gegeben sein soll / muss , muss noch entscheiden werden.

obkram, 2019-10-19:
ok, danke