andreknieriem/photobooth #151: Show date below images in gallery notz working

andi34, 2019-10-20:
**Describe the bug**
White index after enableing the option to show date below images in gallery

**To Reproduce**
Steps to reproduce the behavior:
1. Enable dateformat images
2. enable option to show date below images in gallery
3. save your changes
4. open index
5. only white page

**Expected behavior**
Show index, show date below images in gallery

**Error from console**
```
Fatal error: Uncaught Error: Call to a member function format() on bool in /var/www/html/template/gallery.template.php:16 Stack trace: #0 /var/www/html/index.php(175): include() #1 {main} thrown in /var/www/html/template/gallery.template.php on line 16
```
**Environment (please complete the following information):**
 - OS: Raspbian buster
 - Browser: Chromium Version 74.0.3729.157 (Official Build) Built on Raspbian , running on Raspbian 10 (32-bit)

**Photobooth**
- Did it work before? 
yes

- Which version of Photobooth are you using? 
latest stable

sualko, 2019-10-21:
For me it's not working for freshly taken pictures, but if you reload the page the date is shown (even if it's ugly). Can you confirm this behavior?

[image]

sualko, 2019-10-21:
I added a fix for the ugly date and a check if the filename is in date format. So it looks now like this:

[image]

closed 2019-10-21 by sualko

andi34, 2019-10-21:
Great! Thanks a lot!

commit 7f54ad0f (2019-10-21): fix datetime string on images without date info

fix #151