andreknieriem/photobooth #155: Feature request: updated style-css for font-size of the buttons

msmedien, 2019-10-22:
**Is your feature request related to a problem? Please describe.**
On a screen resolution of 1920x1080 is the button-font-size extremly small and i need to adjust it after every update manually in the css file. :/

**Describe the solution you'd like**
One possible solution is an updated css-file with different media queries for the buttons. You need to adjust the font-size and the margin on the left and right site. Otherwise the button will "explode" in the size and looks to massive.

**Describe alternatives you've considered**
An other solution would be an adjustable font-size (and margin) in the admin panel. - With the second solution everyone can choose the correct value by it self. 

**Additional context**
The main screenshot of this project shows the problem: https://raw.githubusercontent.com/wiki/andreknieriem/photobooth/images/start.png
You need a magnifying glass, if you use a 13" Touchscreen with a resolution of 1920x1080, like me. :P

andi34, 2019-10-22:
@msmedien If you like you can add the media query inside the sassfile and create a pull request with it

sualko, 2019-10-22:
On my 4K screen it looks fine, but I understand the demand for such an option. The easiest thing you can do is to use the zoom option of your browser. Just use ctrl+Plus to enlarge the complete view. The code option would be to completely switch to relative units for font-size. This means we use only `rem` and percentage for the root element. With this solution everyone can set his desired font size in the browser settings.

closed 2020-01-15 by andi34