andreknieriem/photobooth #156: Save data on usb device

kremer8033, 2019-10-22:
**Describe the bug**
I can not save the data folder on an usb device

**To Reproduce**
Steps to reproduce the behavior:
1. Go to admin page
2. Change all folder paths to different subfolder of media/pi/Festplatte/ (e.g. pictures) 
3. save settings
4. See all files in subfolder of var/www/html/media/pi/Festplatte/ (e.g. pictures)

**Expected behavior**
I want that all data files are saved in subfolders of media/pi/Festplatte/ (e.g. pictures). 
So that all data files are saved directly on an usb device

**Environment (please complete the following information):**
 - raspberry pi 3

**Photobooth**
- Did it work before? [no]
- Which version of Photobooth are you using? 2.0.1

In the config file is written under folder paths: media/pi/Festplatte/ (e.g. pictures)

Thanks in advance

andi34, 2019-10-22:
Give proper rights and create a symlink?

kremer8033, 2019-10-22:
I would describe me as an beginner in linux.
Can you give me a short discription or an link with an tutorial.
I use a new system and i only installed the photobooth via the installion routine

andi34, 2019-10-22:
Create a folder called `data` on your usb stick.

Create a symlink inside /var/www/html/data (make sure there's no folder called `data`, gets created automatically while starting Photobooth)
```
ln -s /var/www/html/data /path/to/usb/data
```
Give proper rights to your usb stick for webserver
```
sudo chown -R www-data:www-data /path/to/usb/data
```
Not tested but guess it works

andi34, 2019-10-23:
Forgot to mention: you don't need to adjust the default folder settings following the steps

kremer8033, 2019-10-23:
thanks a lot i will test it and will report.
But are the files than only saved on the usb device or are they saved also on the raspberry pi?
Because i have only less memory capacity left on the raspberry pi.

andi34, 2019-10-23:
If it works it'll only be on the usb stick.

closed 2019-11-05 by andi34