andreknieriem/photobooth #157: Photo Collage Frame

Real2222, 2019-10-23:
**Is your feature request related to a problem? Please describe.**
Resize the Photos and merge with a custom position to a Collage Frame

**Describe the solution you'd like**
Where can i customize and resize the images in the Collage ?

**Describe alternatives you've considered**
Customizable Collage Frame in the Admin Backend

**Additional context**
Like this Preview:
[image]

andi34, 2019-10-23:
Collage creation is done here:
https://github.com/andreknieriem/photobooth/blob/master/lib/collage.php

Print frame is done here:
https://github.com/andreknieriem/photobooth/blob/master/api/print.php#L54
or
https://github.com/andreknieriem/photobooth/blob/master/api/print.php#L88
(Depending on QR print)

sualko, 2019-10-23:
> Where can i customize and resize the images in the Collage ?

That's currently not possible and also currently not on our roadmap.

andi34, 2022-10-04:
Part of Photobooth V4
https://github.com/PhotoboothProject/photobooth/pull/26

closed 2022-10-04 by andi34