andreknieriem/photobooth #158: Change background image in the admin panel

Pfluppe, 2019-10-23:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to change the default background image from the frontend. This would be really helpful because at the moment I always have to change it manually.

**Describe the solution you'd like**
Give an option in the admin panel where you can select a file from the file browser of the device which the webpage is currently loaded on.

If anyone has ideas related to this topic please leave them in this feature request.

Thanks in advance and thank you to everyone contributing!

andi34, 2019-10-23:
Any reason to not fill the template in all parts?

Might be as part of https://github.com/andreknieriem/photobooth/issues/104

Pfluppe, 2019-10-24:
Just saw #57 which is exactly what I want. I'll close this issue.

closed 2019-10-24 by Pfluppe