andreknieriem/photobooth #160: Add display errors option to config

mhellmeier, 2019-10-23:
**Is your feature request related to a problem? Please describe.**
Application always shows error messages like "Gphoto returned with an error code". This is great for analyzing problems but not in a production environment. If you build a photobooth for a wedding and some persons try to make a picture and they get this specific error, they don't know what to do or they get scared (worst case) because they think they broke something.

**Describe the solution you'd like**
Add another option to the config file to enable / display _error display_. Therefore everyone can choose if they want to show the technical related errors to the users or only some standard "I am sorry, something went wrong" clauses.
**Remember**: This suggested _display_error_ config-option should be different from the existing _dev_ option.

andi34, 2019-10-24:
Only that one line you'd like to have changed?
https://github.com/andreknieriem/photobooth/blob/b7113f82d201aa413f22ce09dd3c707a37629750/api/takePic.php#L19

or also 
https://github.com/andreknieriem/photobooth/blob/b7113f82d201aa413f22ce09dd3c707a37629750/api/takePic.php#L26

https://github.com/andreknieriem/photobooth/blob/b7113f82d201aa413f22ce09dd3c707a37629750/api/takePic.php#L63

https://github.com/andreknieriem/photobooth/blob/b7113f82d201aa413f22ce09dd3c707a37629750/api/takePic.php#L71
and
https://github.com/andreknieriem/photobooth/blob/b7113f82d201aa413f22ce09dd3c707a37629750/api/takePic.php#L88

?

mhellmeier, 2019-10-24:
> Only that one line you'd like to have changed?

Not only changing that one line but surrounding the error message output with an if clause that checks weather the _display_error_ config-option is set.

andi34, 2019-10-24:
Possible solution:

[image]

andi34, 2019-10-24:
With this approach it's up to the user what kind of error message appears (you can also add German error messages instead etc.).

mhellmeier, 2019-10-24:
> It was the one error you told, that's why i am asking which errors you like to hide.

I only mentioned one error message to give an example. I don't want to hide only this specific error message, but every error message to reduce the question marks in the faces of the end-users who doesn't understand it.

> And i think it is useful in a production environment to know the error if one shows up.

I have a different opinion in this case (see my example above). That's the reason why I gave the additional _display_error_ config option as a possible solution.

> With this approach it's up to the user what kind of error message appears (you can also add German error messages instead etc.).

I don't think that this is a great solution. It is way better to keep the config file small. If you plan to put every error message into the config, it will blew up the file and make it more complex. Moreover language related output shouldn't be in a config file. If you want to change the language of your system, you need to customize every error message in the config manually. Furthermore every photobooth-admin must update the personal config file every time a new error message will be added.
That's the reason why I suggested to only add **one** additional config parameter. This boolean value determines weather every or no error messages are displayed.

andi34, 2019-10-24:
@mhellmeier I've updated the PR, I'll make it beautiful tomorrow, only mobile right now and not able to test myself today.

andi34, 2019-10-25:
### Option in admin panel
[image]

### disabled error messages
[image]

### enabled error messages
[image]

andi34, 2019-10-27:
@mhellmeier you've tested? Should be fine this way and can be used in other places too if needed.

mhellmeier, 2019-10-27:
Yeah, looks greats! Thats exeactly what I mean and it should be used in the future, too.

So #166  will solve this issue. It's ready to merge, isn't it!? :)

andi34, 2019-10-27:
Yeah, was waiting for your feedback and feedback from sualko, he can merge #166  if he's ok with it too.

sualko, 2019-10-30:
I'm totally against hiding error messages. It results in a bad UX, because the users doesn't know if he caused the failure and someone who has the technical knowledge has no hint why the error occurs. For the non-technical user there is no difference if he sees a error message he doesn't understand, or if there is no message at all. I don't think he will bother with a technical description, it will just complicate troubleshooting.

Everyone has it's own opinion to this topic and while @andi34 already implemented the option, I will merge it asap.

closed 2019-10-30 by sualko