andreknieriem/photobooth #161: Restructure PHP code

mhellmeier, 2019-10-23:
**Is your feature request related to a problem? Please describe.**
The structure of the PHP code is really confusing and different from many other project. The main problem is IMHO that this tool uses a one-function-per-file approach with a lot of includes. Furthermore there are a lot of uncommented functions. This isn't very comfortable and makes it very sluggish for new contributors.

**Describe the solution you'd like**
A better approach would be a class oriented approach with well commented functions and a clear structure.

If you want, I can restructre the PHP part (`api` and `lib` folder) for you in a PR.

sualko, 2019-10-30:
> The main problem is IMHO that this tool uses a one-function-per-file approach with a lot of includes.

That's a result from the history of this project. A couple of weeks ago I tried to enhance and simplify the structure while still maintaining the old style, so that everyone with no major experience in PHP can contribute to this project. If I have started the project, I would use a framework like neos or something similar, but I think that would be to complicated. Of course this is something which can be discussed.

> Furthermore there are a lot of uncommented functions.

Code is documentation, if done right :wink:. If you feel that you don't know what a function is doing it's probably to large or the name was not chosen wisely. But that's only my philosophy.

> If you want, I can restructre the PHP part (api and lib folder) for you in a PR.

I'm fine with another restructuring. Don't know what @andi34 is thinking.

andi34, 2019-10-30:
I am fine with it, we might want to ask @andreknieriem too.
Some parts were moved for better performance on the Pi (main goal to run fine on the Pi).

andreknieriem, 2019-10-30:
You can adjust the code like you want. The complete Code is grown from test to working solution with no rewrite by me. In the past my first version was a website with just one button and a script tag with an ajax call, which calls a php file and makes a shell exec. When this was working, I added the layout and other functions. For me a framework was way to overhead. Nowadays I would build this with some more JS. Maybe vue.js or angular and node with a gphoto library. And I only had some weeks to finish it for my wedding ;) 

So conclusion. If a framework with routing and classes is not worsen the performance, I think it is a good idea to do this. Maybe use https://lumen.laravel.com/ oder symfony in a small version.

mhellmeier, 2019-10-31:
> If I have started the project, I would use a framework like neos or something similar, but I think that would be to complicated. Of course this is something which can be discussed.

> So conclusion. If a framework with routing and classes is not worsen the performance, I think it is a good idea to do this. Maybe use https://lumen.laravel.com/ oder symfony in a small version.

I think a framework with a big model, routing und template engine is way to big for this small project. There are only some small PHP functions in this project. A big part is als JS.

My suggestion for restructuring the PHP code is simply create some classes with static and non-static functions. For example: A class called `Picture` with the static functions `applyFilter(...)`, `resizeImage(...)` and so on. Then, you don't need to include the different one-function-files, the number of PHP files will be reduced and you can simply call `Picture::applyFilter(...)` and so on everywhere without thinking about the correct inclusion.

sualko, 2019-10-31:
> My suggestion for restructuring the PHP code is simply create some classes with static and non-static functions.

Sounds great. If you have time a pr would be appreciated. If you do it, please open it as soon as possible with an estimated completion time, so that we can postbone other changes.

rawbertp, 2019-11-01:
Looking forward to this restructuring as this will also allow to introduce a proper config class/object with getters/and setters for retrieving the properties properly, also allowing to fall back to default where possible and stuff... ultimately getting rid of these 'undefined index...'  :)

sualko, 2019-11-05:
> this will also allow to introduce a proper config class/object with getters/and setters for retrieving the properties properly

We just have to keep in mind, that the config is also used in the js code.

mhellmeier, 2020-01-12:
I just want to let you know that I am currently working on the restructure thing in my fork of this repo (https://github.com/mhellmeier/photobooth/tree/restructurePHP)

I will create a first PR as soon as a meaningful step has reached.

andreknieriem, 2020-01-13:
@mhellmeier Nice!