andreknieriem/photobooth #163: Make collage creation more user friendly

mhellmeier, 2019-10-23:
**Is your feature request related to a problem? Please describe.**
Think of a group of persons who wants to create a photo collage (trying this function the first time). They hit the button, get in position and waiting excited for other instructions. But they will be informed with additional information _after_ the first picture was taken. At this position they realize that only the first picture was made (there are also tools which make 4 pictures in a row) and that a collage consists of 4 pictures. Previously this information was not known.

The main problem is the missing function to cancel the photo collage creating. Let's play the example further: The group leaves after the first picture. Another persons comes to the photobooth. They hit the new image button and wonder after taking 3 pictures why he or she is in a collage with one picture of strangers.

**Describe the solution you'd like**
Create a simple [X] Button in the upper right corner or an orange _Cancel_ button to abort the collage creation.
Perhaps give the users some more information about the collage before taking the first picture. E.g. adding "(4 pictures)" to the button label and describe in a fullscreen message that only the first picture will be taken.

sualko, 2019-10-30:
> there are also tools which make 4 pictures in a row

We have a config option for that.

> Create a simple [X] Button in the upper right corner or an orange Cancel button to abort the collage creation.

PR welcome.

> Perhaps give the users some more information about the collage before taking the first picture.

The icon gives a glue what happens, and I think that users are brave enough to try new things. But if you have a smart idea how this can be enhanced, your pr is appreciated.

rawbertp, 2019-11-10:
I also do not get why one would have to press "New picture" after each and every collage pic. IMO this should be removed as well.

andi34, 2019-11-10:
It's optional and should not be removed.

Cancel button added here https://github.com/andreknieriem/photobooth/pull/193/commits

[image]

rawbertp, 2019-11-10:
Oh... great, it's already there! ;-D

andi34, 2019-11-10:
> Oh... great, it's already there! ;-D

Somehow missed there was a request with a own issue post, thought it was somewhere in the comments on an old PR/Issue.

closed 2019-11-18 by andi34