andreknieriem/photobooth #165: Idea: live view while countdown

DEhrhardt, 2019-10-24:
**Is your feature request related to a problem? Please describe.**

No - it's just a feature request and an idea how to implement a live view.

**Describe the solution you'd like**

I'd like to have a live view directly from my camera. I know that you've already done some investigations to this. I was thinking a lot about possible solutions. My idea for a solution is the following:

When a user or guest clicks on the "New Photo"-Button, a live view will be started and shown while the countdown is on screen. When the white "cheeese"-screen appears, the live view is stopped and the capture-image is triggered.

I think this should be possible with gphoto2 and capture-movie. I don't have an exact solution yet, but maybe you can implement this faster than me. At least I will try to implement this.

**Describe alternatives you've considered**

Alternative solutions I've been thinking about is to use a HDMI capture card (Elgato Cam Link / Auvidea B101 HDMI to CSI-2 Bridge), and connect the live view output of my camera through it, and then use this in combination with motion to get a live view. But this solution is very expensive and means more hardware, interfaces and finally more weak spots.

**Additional context**

Thank you to all contributors, for developing this awesome photobooth solution. I'm new to this and recently started using it these days, I hope I can help to develop this further. 

My name is David, I'm from germany and I'm also an IT-Guy and developer. 😎

andi34, 2019-10-24:
Duplicate https://github.com/andreknieriem/photobooth/issues/98

What I am just doing:
Raspicam is about 25€ on Amazon and works fine at countdown. You only need to place it properly to match your Big Cameras view.
You only need to enable live preview in admin panel to get it working.

closed 2019-10-24 by andi34

andi34, 2019-10-24:
From Andrés comment section:

[image]