andreknieriem/photobooth #167: Pi Camera does not work

chris883773, 2019-10-25:
i have still this problem and get the error:
File was not created.
If i use this
sudo raspistill -n -o $(date +%s) | echo "Done"
at shell i take a image and earn the Message done.

in my.config.inc.php i got the following one:

array ( 'cmd' => 'sudo raspistill -o $(date +%s).jpg -q 100 -t 1 | echo "Done"', 'msg' => 'Done', ), ); i want to use the Raspicam v2 with a touch monitor and a canon printer.

chris883773, 2019-10-25:
Same issue.

andi34, 2019-10-25:
Edit your opening post according to the template and I'll take a look at it and reopen the issue.

andi34, 2019-10-25:
Btw:
```
$config['take_picture']['cmd'] = 'raspistill -n -o $(date +%s) -q 100 -t 1 | echo Done';
$config['take_picture']['msg'] = 'Done';
```
Try those, the " " around the "Done" didn't worked and isn't needed

chris883773, 2019-10-25:
My config looks now like: 

```
<?php
$config = array (
  'take_picture' =>
  array (
    'cmd' => 'raspistill -o $(date +%s).jpg -q 100 -t 1 | echo Done',
    'msg' => 'Done',
  ),
);
```

[image]

I think that the photobooth do not use this cmd.

andi34, 2019-10-25:
I've a pi Cam 2 current at home connected on a Pi4.
I'll find a solution and let you know. Haven't had time to test it for taking pictures, only quick tested with preview at countdown.

Please give me some days.
If you've time you could press F12, try again and send a screenshot of the browser console

chris883773, 2019-10-25:
Ok thank you. i will wait for your solution

andi34, 2019-10-25:
Please try the following:
```
cd /var/www/html/data
sudo -u www-data -s
raspistill -n -o test.jpg -q 100 -t 1 | echo Done
```
Does it work or you're getting any errors?

andi34, 2019-10-25:
Btw. You've followed the actual install instructions mentioned in the wiki too?

andi34, 2019-10-25:
Forget about it, quickly tested and guess I have a solution. I'll try again later and report back.

andi34, 2019-10-25:
@chris883773 problem solved. we changed the permission setup and raspistill does not like it. I am just updating the FAQ, maybe you can report back if it works for you too:

### Is Pi Camera supported?
Yes it is.

First we need to allow our webserver to use `raspistill`. We need add our webserver user to `video` group and reboot once:
```
sudo gpasswd -a www-data video
reboot
```

Next you need to adjust the config:
```
$config['take_picture']['cmd'] = 'raspistill -n -o %s -q 100 -t 1 | echo Done';
$config['take_picture']['msg'] = 'Done';
```
Pi Camera works with these config changes (also works together with preview at countdown if enabled).

chris883773, 2019-10-25:
Okay now it is working. 

But i got now an other problem: 
Cups is not installed in your script. But you need it for printing, is it true?

andi34, 2019-10-26:
Right, Raspbian comes without CUPS.
You'll have to adjust the print cmd.

Type `lpstat -a` in terminal to get the printer name. 
Adjust "$config['print']['cmd'] =" :
```
$config['print']['cmd'] = 'lp -d PRINTERNAME -o landscape -o fit-to-page %s';
```

andi34, 2019-10-26:
Closing the issue as Pi Cam works now.
I only noticed there's some trouble on life view and taking collage at same time, it fails from time to time to take a picture.

closed 2019-10-26 by andi34

Sockenfisch, 2020-06-15:
I got the new RASP CAM HQ for my RASP PI 4 B 4GB.
Unfortunately I can't get any further.

Where do I have to enter the following?
$config['take_picture']['cmd'] = 'raspistill -n -o %s -q 100 -t 1 | echo Done';
$config['take_picture']['msg'] = 'Done';

With me, it comes to the error:
pi@raspberrypi:~ $ $config['take_picture']['cmd'] = 'raspistill -n -o %s -q 100 -t 1 | echo Done';
-bash: [take_picture][cmd]: Command not found.
pi@raspberrypi:~ $ $config['take_picture']['msg'] = 'Done';
-bash: [take_picture][msg]: Command not found.
pi@raspberrypi:~ $

What is the correct command for the Raspberry PI CAM?
Take picture command "gphoto2 --capture-image-and-download --filename=%s"

andi34, 2020-06-15:
First of all:
Enable camera support using the `raspi-config` program you will have used when you first set up your Raspberry Pi.

`sudo raspi-config`

Use the cursor keys to select and open Interfacing Options, and then select Camera and follow the prompt to enable the camera.

No we need to allow our webserver to use `raspistill`. We need add our webserver user to video group and reboot once:
```
sudo gpasswd -a www-data video
reboot
```
Once done open the admin panel in your browser [localhost/admin](http://localhost/admin)

"Take picture command":
`raspistill -n -o %s -q 100 -t 1 | echo Done`

"Success message for take picture":
`Done`

https://github.com/andreknieriem/photobooth/wiki/FAQ#is-pi-camera-supported

Sockenfisch, 2020-06-15:
Aaaarrrg, that's it! Aaaarg I've been looking for this all day!
I am so happy, ... Camera was activated, everything already set up correctly, with only this command missing .... 
"Take picture command":
**raspistill -n -o %s -q 100 -t 1 | echo Done**

Many, many thanks for the quick answer !

andi34, 2020-06-15:
Wiki wasn't 100% up to date here, I've adjusted it here too now.

Good to hear it works now! Hope you enjoy the project!