andreknieriem/photobooth #169: Printing with my Canon Printer does not work

chris883773, 2019-10-26:
Hello,
[image]
My printer does not printing. 
I got the message above and it is doing nothing. 

**My Config**
```
  array (
    'cmd' => 'lp -d Canon_SELPHY_CP1300 -o landscape -o fit-to-page %s',
  ),
```

**My hardware / software**
- Rasperry Pi 4
- PI Camerma v2
- Canon Selphy CP1300
- Touchmonitor
- Chrome Browser

Photobooth
- Newest Version from yesterday
- The printing never worked. A Test site with cups is working

andi34, 2019-10-26:
You've followed the instructions in wiki?

If you like to use the printer you also have to add your webserver user to the lp group:
`sudo gpasswd -a www-data lp`
Now you should restart your Raspberry Pi to apply those settings:
`reboot`

chris883773, 2019-10-26:
Yeah i followed. 
But it does not work. Same problem.

andi34, 2019-10-26:
Check console in browser for errors.

chris883773, 2019-10-26:
That is what i get in console ...Same issue in Firefox and Chrome

```

GEThttp://192.168.178.48/resources/css/style.css
[HTTP/1.1 200 OK 0ms]

GEThttp://192.168.178.48/resources/css/style.css
[HTTP/1.1 200 OK 0ms]

GEThttp://192.168.178.48/node_modules/normalize.css/normalize.css
GEThttp://192.168.178.48/node_modules/normalize.css/normalize.css
GEThttp://192.168.178.48/node_modules/photoswipe/dist/default-skin/default-skin.svg
[HTTP/1.1 200 OK 30ms]

GEThttp://192.168.178.48/resources/fonts/dosis/Regular.woff
GEThttp://192.168.178.48/data/images/20191026_131630.jpg
[HTTP/1.1 304 Not Modified 46ms]

GEThttp://192.168.178.48/data/images/afd9cd64a10e73237408e4e474af201e.jpg
[HTTP/1.1 404 Not Found 80ms]

GEThttp://192.168.178.48/data/images/20191026_131557.jpg
[HTTP/1.1 200 OK 119ms]

XHRGEThttp://192.168.178.48/api/print.php?filename=20191026_131630.jpg
[HTTP/1.1 200 OK 175ms]

GEThttp://192.168.178.48/node_modules/font-awesome/css/font-awesome.css
GEThttp://192.168.178.48/node_modules/photoswipe/dist/photoswipe.css
GEThttp://192.168.178.48/node_modules/photoswipe/dist/default-skin/default-skin.css
[HTTP/1.1 200 OK 0ms]

GEThttp://192.168.178.48/data/images/afd9cd64a10e73237408e4e474af201e.jpg
[HTTP/1.1 404 Not Found 36ms]

XHRGEThttp://192.168.178.48/api/print.php?filename=20191026_131630.jpg
[HTTP/1.1 200 OK 126ms]

Fehler beim Verarbeiten des Wertes für '-webkit-text-size-adjust'.  Deklaration ignoriert. normalize.css:13:29
Unbekannte Eigenschaft '-moz-osx-font-smoothing'.  Deklaration ignoriert. font-awesome.css:20:27
Fehler beim Verarbeiten des Wertes für '-webkit-text-size-adjust'.  Deklaration ignoriert. photoswipe.css:17:29
Unbekannte Eigenschaft '-moz-pointer-events'.  Deklaration ignoriert. default-skin.css:222:23
Fehler beim Verarbeiten des Wertes für '-webkit-perspective'.  Deklaration ignoriert. style.css:88:24
Fehler beim Verarbeiten des Wertes für '-moz-perspective'.  Deklaration ignoriert. style.css:89:21
Fehler beim Verarbeiten des Wertes für 'perspective'.  Deklaration ignoriert. style.css:91:16
Unbekannte Eigenschaft '-moz-filter'.  Deklaration ignoriert. style.css:550:15

​

```

andi34, 2019-10-26:
I don't see an issue here.
You've tried from terminal to print an image using command line?
cd into your images folder using terminal and try to print one as www-data user and see if it works or if there's an error.

To use the command line in terminal as www-data user type:
`sudo -u www-data -s`

chris883773, 2019-10-26:
> Example to print:
lp -d Canon_SELPHY_CP1300 -o landscape -o fit-to-page 20191026_131630.jpg

That printing works.

andi34, 2019-10-26:
As www-data user?

chris883773, 2019-10-26:
EDIT: Only works as PI user. Not as www-data

andi34, 2019-10-26:
> EDIT: Only works as PI user. Not as www-data

What does the terminal say if you try to print that way as www-data user?

andi34, 2019-10-26:
Just wondering about `
```
GEThttp://192.168.178.48/data/images/afd9cd64a10e73237408e4e474af201e.jpg
[HTTP/1.1 404 Not Found 80ms]
```

chris883773, 2019-10-26:
> Just wondering about
> 
> GEThttp://192.168.178.48/data/images/afd9cd64a10e73237408e4e474af201e.jpg
> [HTTP/1.1 404 Not Found 80ms]

It is looking for a picture which was not in my images folder. 

`lp -d Canon_SELPHY_CP1300 -o landscape -o fit-to-page 20191025_232414.jpg
lp: File or folder not found
`
I am in: /var/www/html/data/images/

andi34, 2019-10-26:
`ls -l` Output?

chris883773, 2019-10-26:
```
ls -l
insgesamt 37672
-rw-r--r-- 1 www-data www-data 3678963 Okt 25 23:24 20191025_232414.jpg
-rw-r--r-- 1 www-data www-data 5225212 Okt 25 23:26 20191025_232656.jpg
-rw-r--r-- 1 www-data www-data 5205910 Okt 25 23:30 20191025_233021.jpg
-rw-r--r-- 1 www-data www-data 5548514 Okt 25 23:42 20191025_234257.jpg
-rw-r--r-- 1 www-data www-data 5537221 Okt 25 23:45 20191025_234527.jpg
-rw-r--r-- 1 www-data www-data 5535262 Okt 25 23:54 20191025_235410.jpg
-rw-r--r-- 1 www-data www-data 3995019 Okt 25 23:54 20191025_235427.jpg
-rw-r--r-- 1 www-data www-data 1007208 Okt 26 00:05 20191026_000459.jpg
-rw-r--r-- 1 www-data www-data  984128 Okt 26 10:52 20191026_105248.jpg
-rw-r--r-- 1 www-data www-data  831763 Okt 26 12:34 20191026_123448.jpg
-rw-r--r-- 1 www-data www-data  112770 Okt 26 13:15 20191026_131557.jpg
-rw-r--r-- 1 www-data www-data  890711 Okt 26 13:16 20191026_131630.jpg

```

andi34, 2019-10-26:
Tried again `sudo gpasswd -a www-data lp` and reboot once?

chris883773, 2019-10-26:
Still same Error: 
lp -d Canon_SELPHY_CP1300 -o landscape -o fit-to-page 20191025_232414.jpg lp: File or folder not found

andi34, 2019-10-26:
`/usr/bin/lp Canon_SELPHY_CP1300 -o landscape -o fit-to-page 20191025_232414.jpg`
Does that work?

chris883773, 2019-10-26:
As www-data:

/data/images$ /usr/bin/lp Canon_SELPHY_CP1300 -o landscape -o fit-to-page 20191025_232414.jpg
bash: /usr/bin/lp: File or folder not found

andi34, 2019-10-26:
No idea yet and no printer to test. I let you know if I find out something (if no one's faster).

chris883773, 2019-10-26:
Mh well.... 
Why i can not use the lp as www-data but as normal user ...

andi34, 2019-10-26:
That's what we need to find out...

chris883773, 2019-10-26:
**### let's take a summary**

1. I installed with Noobs Raspian at my PI
2.
```
wget https://raw.githubusercontent.com/andreknieriem/photobooth/master/install-raspbian.sh
sudo bash install-raspbian.sh
```
3. 
```

sudo gpasswd -a www-data video
reboot
```
4.
In my Config
```
$config['take_picture']['cmd'] = 'raspistill -n -o %s -q 100 -t 1 | echo Done';
$config['take_picture']['msg'] = 'Done';
```
5. 
```
sudo gpasswd -a www-data plugdev
sudo gpasswd -a www-data lp
reboot
```

That is what i all doing. I can take pictures but printing does not work with the www-data user. 
With the normal PI user i can do printing in console.

andi34, 2019-10-26:
What I've found, not sure if works to solve the Problem:

"To set up CUPS on Raspberry Pi, you just need to enter a few commands to install the CUPS software, allow printing by other systems, and restart the print server with the new configuration:"
```
$ sudo apt-get install cups
$ sudo cupsctl --remote-any
$ sudo /etc/init.d/cups restart
 ```

chris883773, 2019-10-26:
No effect. Sorry. 

Maype print with lpr? Make this sense?

andi34, 2019-10-26:
Could you tell the output from terminal after typing `which lp` once from pi user and once from www-data user?

chris883773, 2019-10-27:
`pi@fotobox:~ $ lp
^C
pi@fotobox:~ $ sudo -u www-data -s
www-data@fotobox:/home/pi$ lp
lp: Error - /etc/cups/lpoptions file names default destination that does not exist.

www-data@fotobox:/home/pi$ which lp
/usr/bin/lp

www-data@fotobox:/home/pi$ exit
exit

pi@fotobox:~ $ which lp
/usr/bin/lp
`

andi34, 2019-10-27:
`cat /etc/cups/lpoptions` ?
You also might check for user pi and www-data `~/.lpoptions`

chris883773, 2019-10-27:
```
pi@fotobox:~ $ cat /etc/cups/lpoptions
Default Canon_SELPHY_CP1300
pi@fotobox:~ $
pi@fotobox:~ $ sudo -u www-data -s
www-data@fotobox:/home/pi$ cat /etc/cups/lpoptions
Default Canon_SELPHY_CP1300
www-data@fotobox:/home/pi$
```

and lpoptions
```
pi@fotobox:~ $ lpoptions
copies=1 device-uri=gutenprint53+usb://canon-cp1300/B719033109155980 finishings=3 job-cancel-after=10800 job-hold-until=no-hold job-priority=50 job-sheets=none,none marker-change-time=1572101930 marker-colors=#00FFFF#FF00FF#FFFF00 marker-high-levels=100 marker-levels=-3 marker-low-levels=10 marker-message='One or more remaining prints on P media' marker-names=P marker-types=ribbonWax number-up=1 printer-commands=none printer-info='Canon SELPHY CP1300' printer-is-accepting-jobs=true printer-is-shared=true printer-is-temporary=false printer-location printer-make-and-model='Canon SELPHY CP1300 - CUPS+Gutenprint v5.3.1' printer-state=3 printer-state-change-time=1572101930 printer-state-reasons=none printer-type=135244 printer-uri-supported=ipp://localhost/printers/Canon_SELPHY_CP1300

www-data@fotobox:/home/pi$ lpoptions
www-data@fotobox:/home/pi$

```

andi34, 2019-10-29:
@chris883773 maybe next month I get a printer and can test myself too.
Does `~/.lpoptions` exist as www-data user? If yes you could try if removing it solves your problem.

Else:
`lp` should use the default printer defined in `/etc/cups/lpoptions` if I am not wrong.
What happens if you run
`lp -o landscape -o fit-to-page 20191025_232414.jpg`
? Does it work from both user (pi and www-data) ?

andi34, 2019-10-30:
@sualko you've a printer an can test maybe?

sualko, 2019-10-31:
At least on my Ubuntu laptop the user www-data is able to print. If I scroll through the issue most of the commands are wrong. First you have to check if lp recognizes your printer. Type `lpstat -a` and there should be a list like:

```
_HP-PageWide-Pro-Buero accepting requests since Do 31 Okt 2019 15:15:33 CET
Brother_DCP_J152W_Sarahs_MacBook_Pro accepting requests since Di 08 Mai 2018 09:32:41 CEST
Brother_DCP_J152W_Sarahs_MacBook_Pro@Sarahs-MacBook-Pro-5.local accepting requests since Mi 30 Mai 2018 12:11:52 CEST
Brother_DCP_J152W_Sarahs_MacBook_Pro@Sarahs-MacBook-Pro-6.local accepting requests since Mi 27 Jun 2018 12:06:05 CEST
...
```

Now you pick your desired printer, in my example `_HP-PageWide-Pro-Buero`, and try to print a simple text file (it should exist and not be empty, e.g. `echo Hello > example.txt`):

```
lp -d _HP-PageWide-Pro-Buero example.txt
```

If this works, try it as www-data: `sudo -u www-data lp -d _HP-PageWide-Pro-Buero example.txt`.

If this also works try to create your image print command. It should be something like `lp -d _HP-PageWide-Pro-Buero -o orientation-requested=5 -o fit-to-page IMAGE_NAME`.

Real2222, 2019-10-31:
I got the same issue, the canon printer would not recognize.

`pi@Photobooth:~ $ lpstat -a
(null) akzeptiert keine Anfragen seit Do 31 Okt 2019 19:47:10 CET -
        reason unknown
Brother_MFC-L2700DN_series_Printer_über_VNC_von_DESKTOP-VKO4TCF akzeptiert anfragen seit Do 31 Okt 2019 19:47:44 CET
Brother_MFC_L2700DN_series akzeptiert anfragen seit Do 31 Okt 2019 19:47:38 CET
`
`pi@Photobooth:~ $ lsusb
Bus 001 Device 005: ID 1997:2433
Bus 001 Device 008: ID 04a9:3214 Canon, Inc. SELPHY CP800
Bus 001 Device 007: ID 04a9:3148 Canon, Inc. PowerShot S5 IS
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

Real2222, 2019-11-01:
> 
> 
> ```
> pi@fotobox:~ $ cat /etc/cups/lpoptions
> Default Canon_SELPHY_CP1300
> pi@fotobox:~ $
> pi@fotobox:~ $ sudo -u www-data -s
> www-data@fotobox:/home/pi$ cat /etc/cups/lpoptions
> Default Canon_SELPHY_CP1300
> www-data@fotobox:/home/pi$
> ```
> 
> and lpoptions
> 
> ```
> pi@fotobox:~ $ lpoptions
> copies=1 device-uri=gutenprint53+usb://canon-cp1300/B719033109155980 finishings=3 job-cancel-after=10800 job-hold-until=no-hold job-priority=50 job-sheets=none,none marker-change-time=1572101930 marker-colors=#00FFFF#FF00FF#FFFF00 marker-high-levels=100 marker-levels=-3 marker-low-levels=10 marker-message='One or more remaining prints on P media' marker-names=P marker-types=ribbonWax number-up=1 printer-commands=none printer-info='Canon SELPHY CP1300' printer-is-accepting-jobs=true printer-is-shared=true printer-is-temporary=false printer-location printer-make-and-model='Canon SELPHY CP1300 - CUPS+Gutenprint v5.3.1' printer-state=3 printer-state-change-time=1572101930 printer-state-reasons=none printer-type=135244 printer-uri-supported=ipp://localhost/printers/Canon_SELPHY_CP1300
> 
> 
> www-data@fotobox:/home/pi$ lpoptions
> www-data@fotobox:/home/pi$
> ```
> 
> It looks like as www-data can not use lpoptions

This works for me.
Can you try this command?
sudo usermod -a -G lpadmin www-data

chris883773, 2019-11-02:
> Can you try this command?
sudo usermod -a -G lpadmin www-data

After trying this one. I can print. 

But if the printer has no more paper if is going in holding. And comes not back from alone.

flighter18, 2019-11-04:
@chris883773 
 > But if the printer has no more paper if is going in holding. And comes not back from alone.

I Have the same printer and the same problem.
My solution at the moment is a new subfolder "/cups/ with a short script, which enables the printer after pushing the  "start printer" button.

[image]

chris883773, 2019-11-04:
> @chris883773
>
>
> I Have the same printer and the same problem.
> My solution at the moment is a new subfolder "/cups/ with a short script, which enables the printer after pushing the "start printer" [...]

Can you give me your script? :) Than we can do it at index for the customers

flighter18, 2019-11-04:
Here you go :

[printer_action.sh.txt](https://github.com/andreknieriem/photobooth/files/3803606/printer_action.sh.txt)

You also need to add the following line to suoders:

`www-data  ALL=(root) NOPASSWD: /var/www/html/cups/printer_action.sh`

chris883773, 2019-11-04:
Maybe after printing:
Doing like a cronjob and checking the error page of printer. 
So the customers get a popup if is anything to do?

flighter18, 2019-11-04:
unfortunately the error handling of printer / cups is not that good. 

If the printer runs out of paper, there will be a message after a while. But the message does not appear immediately and this is a problem: 

Just think of: while taking a new collage sudenly the Printer PopUp "please refil the paper" appears. That wouln't be good.

My idea is to get a sencond touchscreen.

chris883773, 2019-11-04:
> Just think of: while taking a new collage sudenly the Printer PopUp "please refil the paper" appears. That wouln't be good.

After he is printing in the script, "Waiting for print" --> "please refil the paper".

flighter18, 2019-11-04:
> > Just think of: while taking a new collage sudenly the Printer PopUp "please refil the paper" appears. That wouln't be good.
> 
> After he is printing in the script, "Waiting for print" --> "please refil the paper".

Yeah but you could interrupt the next shot.

Most of the time, the queue hasn't just one pic. The printer prints one by one. 
So the printer could run out of papier during the second job, long after you pushed the "print button". you can't get the feedback from the printer immediately

chris883773, 2019-11-08:
Ok than i think it is the best if you can do your script at index.php. So the customer can reset him after refil paper.

andi34, 2019-11-08:
https://github.com/andreknieriem/photobooth/wiki/Installation-on-Debian/_compare/0e19624864f0ddc5e4791dbdb6c3a95da21c7cbd...b880a3673803a71dd5f2d608318a82eed46b3ba7

Added to wiki. Giving root access to a bash file isn't a solution from security point of view (for your 2nd problem) and not related to this topic.

Not sure if there's a way to reset the printer que via `lp` command line, if yes there's no trouble to implement a Button. Feel free to open a proper feature request.

closed 2019-11-08 by andi34