andreknieriem/photobooth #172: Poor image quality

rawbertp, 2019-11-01:
Hi!

First of all thanks for your work on this awesome project! :)

And now to my problem/question: The jpgs in the data folder only are around 600K in size (whereas they should be ~3M). I haven't found any setting that would allow me to set the compression but it appears that during image processing the JPGs are compressed/whatever.

Would it be possible to introduce some configuration options that allow the user to control/turn off this behavior?

Thanks,
Robert

andi34, 2019-11-01:
Please edit your opening post using the bug or feature template, will be reopened once the opening post is updated.

rawbertp, 2019-11-01:
Is it valid now?

andi34, 2019-11-01:
data/images or data/tmp ?

rawbertp, 2019-11-01:
It's about `data/images` (I updated the original post accordingly). It appears the original JPGs are in the `data/tmp` folder. However, to me a `tmp` folder is something rather short-lived?!

andi34, 2019-11-01:
tmp folder was introduced to store the original files before they get resized for chroma keying and to store the original picture before a filter is applied.

rawbertp, 2019-11-01:
Alright, thanks for that. So maybe it could be renamed to `original` or something like that which would be far less confusing. Only a cosmetic issue, though.

However, as I have not enabled/used any filters, why is the image in `images` 1/5 the size of the original? And even when applying filters it would be nice to have an option to set the compression level/quality loss. :)

andi34, 2019-11-01:
You can adjust the quality here if needed:
https://github.com/andreknieriem/photobooth/blob/master/api/applyEffects.php#L89

Change from 80 to 100.

andi34, 2019-11-01:
> Alright, thanks for that. So maybe it could be renamed to `original` or something like that which would be far less confusing. Only a cosmetic issue, though.
> 

You can adjust the folder name via config/admin panel already.

rawbertp, 2019-11-01:
> You can adjust the quality here if needed:
> https://github.com/andreknieriem/photobooth/blob/master/api/applyEffects.php#L89
> 
> Change from 80 to 100.

Thanks for this hint, will do!

> You can adjust the folder name via config/admin panel already.

Yeah, you're right! I haven't considered that as `tmp` to me is not a directory for persistent files. Now I know better...

rawbertp, 2019-11-01:
Interestingly, when changing `imagejpeg($imageResource, $filename_photo, 80);` to `imagejpeg($imageResource, $filename_photo, 100);` the JPG in `images` is now twice the size of the original in `tmp`. How does that make sense again?

andi34, 2019-11-01:
GD is used for image processing here. it is creating a new image from the original file which is used by Photobooth. it's not a real 1:1 copy like copy and paste.

andi34, 2019-11-01:
You might want to take a look here too: 
https://github.com/andreknieriem/photobooth/issues/117

sualko, 2019-11-01:
> You can adjust the folder name via config/admin panel already.

That doesn't solve the issue and is dangerous. Assume someone implements a garbage collector for the temp folder, than all your images are gone.

andi34, 2019-11-01:
> > You can adjust the folder name via config/admin panel already.
> 
> That doesn't solve the issue and is dangerous. Assume someone implements a garbage collector for the temp folder, than all your images are gone.

If it's dangerous why we allow to change the path via config / admin panel?

rawbertp, 2019-11-01:
The question is, what is the main purpose of this folder? From what I've seen so far in the code, it doesn't really serve as a **temporary** folder as such. It's rather a permanent storage for the original files and hence should be probably renamed to reflect its actual purpose.

andi34, 2019-11-03:
Maybe we can think about renaming it to `original` in the next major version bump @sualko ?
```
api/applyEffects.php:$filename_tmp = $config['foldersAbs']['tmp'] . DIRECTORY_SEPARATOR . $file;
api/applyEffects.php:    $collageBasename = substr($filename_tmp, 0, -4);
api/applyEffects.php:    if (!createCollage($collageSrcImagePaths, $filename_tmp)) {
api/applyEffects.php:if (!file_exists($filename_tmp)) {
api/applyEffects.php:$imageResource = imagecreatefromjpeg($filename_tmp);
api/takePic.php:$filename_tmp = $config['foldersAbs']['tmp'] . DIRECTORY_SEPARATOR . $file;
api/takePic.php:    takePicture($filename_tmp);
api/takePic.php:    $basename = substr($filename_tmp, 0, -4);
config/config.inc.php:$config['folders']['tmp'] = 'data/tmp';
lib/configsetup.inc.php:                'tmp' => [
lib/configsetup.inc.php:                        'placeholder' => 'tmp',
lib/configsetup.inc.php:                        'name' => 'folders[tmp]',
lib/configsetup.inc.php:                        'value' => $config['folders']['tmp']
resources/js/core.js:        const tempImageUrl = config.folders.tmp + '/' + result.file;
resources/lang/de.js:    'folders_tmp': 'tmp Ordner',
resources/lang/en.js:    'folders_tmp': 'tmp Folder',
resources/lang/es.js:    'folders_tmp': 'Carpeta búfer',
resources/lang/fr.js:    'folders_tmp': 'Dossier tmp',
resources/lang/gr.js:    'folders_tmp': 'tmp φάκελο',
```
Not much lines to adjust.

And yeah, somehow it is a temporary folder as it is used for the original files before they get adjusted for the booth. No user can see them on the webpage.

We could also think about adding an option to keep or remove the original files.

sualko, 2019-11-05:
> Maybe we can think about renaming it to original in the next major version bump @sualko ?

That would be an option, but the name should always reflect the use of it. I think we can also skip the jpeg conversion, if no filters/borders and so on are applied. That would also enhance the performance.