andreknieriem/photobooth #177: Prefer nginx over Apache httpd / nginx instructions

rawbertp, 2019-11-02:
**Is your feature request related to a problem? Please describe.**
No problem but I suggest to replace Apache httpd installation instructions/requirements by nginx. Nginx has a smaller memory footprint and typically better performance, which is especially important on the Raspberry Pis.

**Describe the solution you'd like**
1. Move the installation instructions from the Wiki to the `README.md`. This would allow others to contribute.
2. Describe how to install/configure `nginx` and `php-fpm`. This is quite straight-forward and a task of 2-5 minutes. :)

**Describe alternatives you've considered**

One could also provide a howto for both httpd and nginx but I would suggest to make nginx default for reasons mentioned above.

**Additional context**

andi34, 2019-11-02:
I'd suggest keeping the instructions in the wiki.
We moved everything out of the README for a better documentation.
If you like you can tell the instructions here and we can add it to the wiki if a decision is made.

andi34, 2019-11-03:
Could look like this in the wiki.

### Install a Webserver
Currently Apache2 and NGINX Webserver are supported. NGINX is more lightweight, but needs some additional steps until you're good to go.

#### Install Apache2 & PHP
```
sudo apt install -y libapache2-mod-php
```

#### Install NGINX & PHP
```
sudo apt install -y nginx php-fpm
```

Once NGINX is installed we need to enable PHP in NGINX:
```
sudo nano /etc/nginx/sites-enabled/default
```
Find the line `index index.html index.htm;` add add `index.php` after `index` (the line now should look like this: `index index.php index.html index.htm;`).

Scroll down until you find a section with the following content:
```
# pass the PHP scripts to FastCGI server listening on 127.0.0.1:9000
#
# location ~ \.php$ {
```

Edit by removing the `#` characters on the following lines:
```
location ~ \.php$ {
    include snippets/fastcgi-php.conf;
    fastcgi_pass unix:/var/run/php5-fpm.sock;
}
```
It should look like this:
```
        # pass the PHP scripts to FastCGI server listening on 127.0.0.1:9000
        #
        location ~ \.php$ {
                include snippets/fastcgi-php.conf;

        # With php-fpm (or other unix sockets):
        fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
        # With php-cgi (or other tcp sockets):
    #    fastcgi_pass 127.0.0.1:9000;
        }
```

Now we need to start the server with:
```
sudo /etc/init.d/nginx start
```

### Install dependencies
```
sudo apt install -y git php-gd gphoto2
```

andi34, 2019-11-03:
Quick tested with nginx on a fresh Raspbian Buster image, followed the instructions above. Works fine so far.

andi34, 2019-11-03:
Added to the instruction:
https://github.com/andreknieriem/photobooth/wiki/Installation-on-Debian/_compare/6accec5fe65e654b229c75e421f6d5a88149789c...be6f075ae317836a31b671755b55ee136822b1ec

Any idea for the installation script to use NGINX by default?

rawbertp, 2019-11-03:
The following sed commands will uncomment the corresponding lines:

```bash
sed -i '/location ~ \\.php$ {/s/^\(\s*\)#/\1/g' /etc/nginx/sites-available/default
sed -i '/include snippets\/fastcgi-php.conf/s/^\(\s*\)#/\1/g' /etc/nginx/sites-available/default
sed -i '/fastcgi_pass unix:\/run\/php\//s/^\(\s*\)#/\1/g' /etc/nginx/sites-available/default
sed -i '/.*fastcgi_pass unix:\/run\/php\//,// { /}/s/^\(\s*\)#/\1/g; }' /etc/nginx/sites-available/default
```

andi34, 2019-11-04:
Thanks :) I'll see if I find time in the evening to test it too

rawbertp, 2019-11-04:
One thing probably worth mentioning - this has only been tested on the happy path, i.e. a vanilla nginx installation (on Debian Buster). It may not work/cause issues if the user already did manual changes himself or for other distributions which ship with another config (structure).

andi34, 2019-11-07:
NGINX,
Applied Antique Filter
```
{success: "image", file: "cb9517914ba5c9f5737994ca610b9881.jpg", time: 2.4943900108337402}
core.js:257 picture processed 
{file: "cb9517914ba5c9f5737994ca610b9881.jpg", time: {…}}
file: "cb9517914ba5c9f5737994ca610b9881.jpg"
time:
chroma: 0.7251811027526855
filter: 2.015481948852539
image: 0.46349191665649414
polaroid: 0.0000059604644775390625
thumb: 0.23497796058654785****
```

andi34, 2019-11-07:
there's a small difference, even if it is not that big.
on both the first picture takes some more time, while every picture after that is a little faster done.

rawbertp, 2019-11-07:
Thanks for performing these tests. What would be also/especially interesting are memory (RAM) and CPU utilization.

andi34, 2019-11-07:
Ordered 2 more MicroSD cards so I don't have to format the same card many times a day ;)

Can tell more next days.

closed 2019-11-11 by sualko