andreknieriem/photobooth #178: delete a session

mpler, 2019-11-02:
I find it awkward the way over the file manager to delete files

how about a delete function in admincp

the photos would still be on the memory card but in the gallery would be only photos of the current event

sualko, 2019-11-05:
There is already a delete option on the admin page, but it deletes also all files. If you like you can create a pull request to add a nice dialog which allows to choose what should be deleted.

mhellmeier, 2019-11-06:
There is also a reset script, implemented in #171

andi34, 2019-12-30:
Merged in #206 

- Allow admins to choose what gets deleted at reset (inspired by https://github.com/andreknieriem/photobooth/issues/178)
  - always:
    - delete db.txt
  - optional (but enabled by default):
    - delete images
    - delete "mail-addresses.txt
    - delete personal config (my.config.inc.php)

closed 2019-12-30 by andi34