andreknieriem/photobooth #182: Guest access for viewing/downloading pictures

rawbertp, 2019-11-05:
**Is your feature request related to a problem? Please describe.**

When using PhotoBooth in a "hotspot-environment" it should be possible to differentiate between devices with elevated privileges (typically "localhost" and/or a tablet used in the booth) and "guest devices" (all other devices accessing PB). The latter ones should only have access to the gallery. They must not be able to take or print but only view and download photos (this would also required #7 ).

**Describe the solution you'd like**

Create a new config option à la `privileged_devices` defaulting to `127.0.0.1`. Multiple devices can be added.

* Privileged devices will have access to all functions (as is).
* All the other (guest) devices will only have access to the gallery (`gallery.php` should be the landing page for them).
* This would also be a first measure to secure the admin area from unauthorized access.

**Describe alternatives you've considered**

**Additional context**

andi34, 2019-11-05:
Maybe also aside https://github.com/andreknieriem/photobooth/pull/118

It's nothing I am currently able to to... Missing knowledge/skills in that direction

rawbertp, 2019-11-05:
Yeah, but TBH I do not consider a login/authentication feature that important for this software. Or let me put it another way: Currently PB is not meant to be publicly exposed (i.e. reachable from the internet). If this was a requirement, not only authentication/authorization would have to be added but a general auditing/hardening would be required. I'm pretty sure that plenty of exploits/attack vectors could be successfully applied to the current state of the code.

I personally, do not consider that a big deal, as long as it is clearly stated, that it is meant to be used in an offline environment only. Assuming that, it should be sufficient to restrict the admin area to certain IPs (at a first step) which should also be quite straight-forward to implement.

andi34, 2019-11-05:
Adding a check to check for localhost or defined IP, if not match open gallery.php ?

sualko, 2019-11-05:
This can also be done via a webserver config, as mentioned in #118 and I think this is a duplicate of #39, therefore I close this issue.

closed 2019-11-05 by sualko

rawbertp, 2019-11-05:
Sorry, but I do not agree that this is a duplicate of #39 and the webserver config only solves part of the problem.

The use case I'm referring to/want to solve with this feature request is to allow guests to access PB via their mobile devices and download pictures. But obviously they must not be allowed to print, take a picture or access the admin area.

rawbertp, 2019-11-05:
Yes and no. :-) The login authentication fix would **not** solve the problem that _remote users_ can _print_ and _take pictures_. Unless the "master" device has to authenticate too but I don't think that someone wants to login the iPad installed in the booth. ;-) Hence, I believe that doing this on an IP level is the easiest approach and security wise it should be sufficient unless "being online" gets an requirement.

sualko, 2019-11-05:
We could also leave this open, but I think in both cases it's about authorization and if you fix one, you automatically fix the other. Hope this is ok for you.