andreknieriem/photobooth #185: Apple devices smaller IOS 13.x / Gallery Fullscreenpicture button without function

obkram, 2019-11-06:
**Describe the bug**
Bei all meinen Appledevices ( iPhone / iPad ) mit einem IOS kleiner ios 13.0 funktionieren die Funktionsbutton in der ( Vollbild ) Gallery Ebene nicht, wenn ein Bild sich im Fullscreen Modus befindet. Die Icons zeigen zwar an das sie betätigt werden ( werden weiss ) aber es folgt keine Aktion. Getestet mit Chromakeying, Printing und Barcode. Nur die "X" Taste zum zurückkehren in die übergeordnete Galleryebene funktioniert.
Testet man den Barcode aus einer anderen Ebene z.B. direkt nach der Bildaufnahme dann funktioniert dies.
Ich habe extra einen aktuelle Browser eines anderen Herstellers ( Firefox etc. ) ausprobiert. Dort ging es ebenfalls nicht. Nur an einer älteren Version von Safari allein kann es nicht liegen.
Ich weiss das der Fehler schonmal als Issue eröffnet wurde aber ich habe den Issue nicht in den Offenen Issues finden können.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Gallery, via APPLE iPad/iPhone smaller IOS13'
2. Click on 'one of the pictures'
3. Click on 'chromakeying or print or QR Code, they don´t work. Only color will be changed shortly'
4. Click on 'X, this will work'

**Expected behavior**
Die entsprechende Funktion sollte nach dem Antippen ausgeführt werden. Die Icons zeigen die Auslösung via Farbwechsel an aber nichts passiert.

**Screenshots**
nicht möglich, Format wird nicht akzeptiert

**Environment (please complete the following information):**
 - OS: IOS 12.4.3 ( not working )
 - OS: IOS 13.2 ( working )
 - Browser - Safari stand IOS 12.4.3/ Firefox aktuell

**does it work bevore**
- Viel früher schon. Habe seit dem auf IOS13 oder direkt auf dem Browser des PI getestet wo es funktionierte.

**Photobooth**
- absolut aktuell. Läuft auf einen RPI3B auf Raspbian Buster.

Upload einer Videos war leider nicht möglich. Das Format wird von der Seite nicht akzeptiert.
Als Betatester stehe ich gerne mit der entsprechenden Hardware zur Verfügung. Vielen Dank.

flighter18, 2019-11-06:
Hallo @obkram 

Dieser Bug ist bereits bekannt. Ich hatte das selbe Problem, siehe https://github.com/andreknieriem/photobooth/issues/116 & https://forum.webflow.com/t/button-not-responding-in-safari-on-iphone/51418.

Bitte beachte, dass dies ein open source Projekt ist und somit nicht genug Ressourcen zur Verfügung stehen, um alle alten Betriebssysteme unterstützten zu können.

andi34, 2019-11-06:
@sualko https://stackoverflow.com/questions/15280596/css-div-buttons-not-working-on-iphone-or-ipad might be a solution?

sualko, 2019-11-07:
I don't own any Apple products and have no time to test other environments. If you create a pull request to fix this issue, I'm happy to test and comment it.

andi34, 2019-11-07:
@flighter18 could you test the following please:
```
diff --git a/resources/js/photoinit.js b/resources/js/photoinit.js
index 81e7566..43489a9 100644
--- a/resources/js/photoinit.js
+++ b/resources/js/photoinit.js
@@ -135,8 +135,7 @@ function initPhotoSwipeFromDOM (gallerySelector) {
     };
 
     // QR in gallery
-    $('.pswp__button--qrcode').on('click', function (e) {
-        e.preventDefault();
+    $('.pswp__button--qrcode').on('click', function () {
 
         const pswpQR = $('.pswp__qr');
 
@@ -154,8 +153,7 @@ function initPhotoSwipeFromDOM (gallerySelector) {
     });
 
     // print in gallery
-    $('.pswp__button--print').on('click', function (e) {
-        e.preventDefault();
+    $('.pswp__button--print').on('click', function () {
 
         const img = gallery.currItem.src.split('/').pop();
 
@@ -165,8 +163,7 @@ function initPhotoSwipeFromDOM (gallerySelector) {
     });
 
     // chroma keying print
-    $('.pswp__button--print-chroma-keying').on('click', function (e) {
-        e.preventDefault();
+    $('.pswp__button--print-chroma-keying').on('click', function () {
 
         const img = gallery.currItem.src.split('/').pop();
 
```

andi34, 2019-11-07:
Forget about it. From what I found:

"The problem with .click on iOS is that it unbinds the click event."

"(...) using 'click touchstart' will get the desired result. If you console.log(e) your clicks though, you may find that when jquery recognizes touch as a click - you will get 2 actions from click and touchstart. The solution bellow worked for me.
```
//if its a mobile device use 'touchstart'
if( /Android|webOS|iPhone|iPad|iPod|BlackBerry|IEMobile|Opera Mini/i.test(navigator.userAgent) ) {
    deviceEventType = 'touchstart'
} else {
//If its not a mobile device use 'click'
    deviceEventType = 'click'
}

$(document).on(specialEventType, function(e){
    //code here
});
```
"

flighter18, 2019-11-08:
@andi34 OK, I've forgotten it :)

andi34, 2019-11-08:
@flighter18 could you please test https://github.com/andi34/photobooth/commit/8c62f1c6bb22bf211ca53feb80cbd3ad84d5e608 ?

flighter18, 2019-11-08:
@andi34 i will test it this weekend.

obkram, 2019-11-08:
@andi34 Ich bin zwar nicht flighter18 aber ich hatte bereits Zeit euren Fix zu testen. Funktioniert ( zumindest bei mir ) einwandfrei. Habt ihr super hinbekommen, vielen Dank.

andi34, 2019-11-08:
@obkram danke :)
Werde es nachher per PR hier reinpacken, dann kann @sualko es nochmal absegnen.

andi34, 2019-11-08:
Maybe someone with a non-apple device and printer could check that print isn't executed twice if clicking the print button in gallery after this change?

andi34, 2019-11-10:
There's a seperate issue on iOS 12 and below with QR and if implemented DL Button.

Todo: detect if we're using an iOS devices, if yes also check the version and spit an error message that iOS is to old to use this function.

Might help on implementing the check:
https://stackoverflow.com/questions/8348139/detect-ios-version-less-than-5-with-javascript

obkram, 2019-11-10:
> Maybe someone with a non-apple device and printer could check that print isn't executed twice if clicking the print button in gallery after this change?

Habe es mit einen Firetablet getestet. Es gab nur einen Ausdruck.

flighter18, 2019-11-10:
> @flighter18 and @obkram could you please test [andi34@8c62f1c](https://github.com/andi34/photobooth/commit/8c62f1c6bb22bf211ca53feb80cbd3ad84d5e608) ?
> 
> I'll fix the Apple typo in commit message once confirmed it fixes the issue

sorry had a busy weekend.
It didn't work on my IPad mini 2 with IOS 12.4.2.

flighter18, 2019-11-10:
> @flighter18 and @obkram could you please test [andi34@8c62f1c](https://github.com/andi34/photobooth/commit/8c62f1c6bb22bf211ca53feb80cbd3ad84d5e608) ?
> 
> I'll fix the Apple typo in commit message once confirmed it fixes the issue

sorry had a busy weekend :)
Works prefectly with my IPad 2 and IOS12. Get also just one print.

andi34, 2019-11-10:
Thanks @flighter18 and @obkram

andi34, 2019-11-11:
> There's an seperate issue on iOS 12 and below with QR and if implemented DL Button.
> 
> Todo: detect if we're using an iOS devices, if yes also check the version and spit an error message that iOS is to old to use this function.
> 
> Might help on implementing the check:
> https://stackoverflow.com/questions/8348139/detect-ios-version-less-than-5-with-javascript

https://stackoverflow.com/q/57599945

Looks like we shouldn't check for the iOS version

closed 2019-11-14 by sualko