andreknieriem/photobooth #188: Button Border Radius

obkram, 2019-11-07:
**Is your feature request related to a problem? Please describe.**
Ich habe bei mir heute eine Einstellung ausprobiert um die Button etwas weicher erscheinen zu lassen. Es betrifft alle Button Ausnahme des Button X und dem Reset Button im Admin Panel. Soweit habe ich keine Probleme dadurch feststellen können.

**Describe the solution you'd like**
Nach Änderung der Style.css erscheinen die Button etwas weicher und nicht ganz so kantig. 

**Describe alternatives you've considered**
Zur Umsetzung habe ich in der Style.css unter die Zeile 108 noch folgenden Parameter angefügt:
....
border-radius: 10px; }

**Additional context**
Vielleicht gefällt euch das ja auch.

andi34, 2019-11-09:
@obkram i've applied your change to `resources/sass/partials/_button.scss ` 
https://github.com/andi34/photobooth/commit/0cb1141a5d39d197c4bf7d50cd6aeed93571943c

[image]

[image]

[image]

I've not touched the reset and save button in admin panel, does not look that good imo:

[image]

[image]

obkram, 2019-11-09:
@andi34 Vielen Dank für die Integragtion.

andi34, 2019-11-09:
Nur auf meinem fork in einem branch ;)

obkram, 2019-11-09:
@andi34 was Branches, Cherry Picks und solche Sachen angeht da bin ich ehr ein absoluter Muggel, aber es klappt doch immer irgendwie ;). Schönes Wochenende zusammen.

andi34, 2019-11-10:
Round edges in Admin Panel:
https://github.com/andi34/photobooth/commit/2979af0ea64818127eadbb1bdbb31d6283886985

[image]

andi34, 2019-11-10:
https://github.com/andi34/photobooth/commit/15bdc244eff3621b35a249ade301ce0efa5d0872

[image]

obkram, 2019-11-10:
@andi34 Das sieht großartig aus!

andi34, 2019-11-10:
And some space between the buttons on the result page:
https://github.com/andi34/photobooth/commit/94f6e2f0f638d5723dfeca48c4910565ae8595e3

[image]

andi34, 2019-11-10:
Adjusted Admin Panel & close/reload button in Gallery

https://github.com/andi34/photobooth/commit/3c1ef3ae510ba04658817ae2c604e23f8c2d338c

https://github.com/andi34/photobooth/commit/9175b1698c08c7b99f1f16eb824378705490e346

[image]

[image]

[image]

andi34, 2019-11-10:
**Look on mobile device (OnePlus 6)**

[image]

[image]

[image]

[image]

[image]

andi34, 2019-11-13:
Some adjustments to the Gallery button, looks better on the rounded corners:

[image]

sualko, 2019-11-14:
I'm not a fan of round corners, but if that's common sense we can merge it anyway. One remark: Can you try to make only the outer corners round? So have no border radius at the edges between two buttons. Also only a round corner on the bottom right corner of the gallery button. Just an idea as kind of a compromise :-).

andi34, 2019-11-14:
@sualko not sure if we should merge it. Just kept it here for other people.

Since I am learning css still, can you tell how to make only one side rounded?

rawbertp, 2019-11-14:
I'm not a big fan of rounded corners/buttons too... Didn't all major UIs get rid of them in favour of flat, rectangular buttons/tiles? Anyways... if they have to be rounded, maybe the following can be used as inspiration: https://mdbootstrap.com/docs/angular/components/buttons/

andi34, 2019-11-14:
Don't worry, it's not a must have. I've just played around with what @obkram started with.

Maybe someone has an idea on how to make it an config option?

sualko, 2019-11-21:
> can you tell how to make only one side rounded?

https://developer.mozilla.org/en-US/docs/Web/CSS/border-radius

> Maybe someone has an idea on how to make it an config option?

We could ship different templates (css files) and add a switch to choose which css file should be used. For me this is no priority.

closed 2020-01-03 by andi34