andreknieriem/photobooth #190: Filter after taking photo

chris883773, 2019-11-08:
Hi,

filter changes of the picture after doing this. 

**For example**
[image]

**Procedure**
Take a picture
Choose a filter
Save picture as "normal" and with filter (maybe you can choose this at config)
Adding picture at collage or at normal photos
Optional: Printing per button

Have a nice day!

andi34, 2019-11-08:
Waiting for your Pullrequest. :)
Closing since duplicate https://github.com/andreknieriem/photobooth/issues/96

closed 2019-11-08 by andi34

andi34, 2019-11-08:
> Save picture as "normal" and with filter (maybe you can choose this at config)

Btw. If a filter is applied the original image stays in /data/tmp but isn't used by the gallery

rawbertp, 2019-11-10:
> Closing since duplicate #96

Is this really a dup? In #96 @sualko mentioned to show a(ny) preview. The request here, like my comment in #96, is to apply the filter to the actual photo.

andi34, 2019-11-10:
Since the general topic in https://github.com/andreknieriem/photobooth/issues/96 is to enhance the filter implementation in the whole process I'd say it is a duplicate. Can be reopened if I am wrong.

chris883773, 2019-11-11:
> > Closing since duplicate #96
> 
> Is this really a dup? In #96 @sualko mentioned to show a(ny) preview (to my understanding this could be just a random sample picture). The request here, like my comment in #96, is to apply the filter to the actual photo.

That is true. I want to take a picture. And after taking a picture choosing the filter. 
Than take a new picture with button.

rawbertp, 2019-11-15:
@chris883773 You might want to have a look at the linked PR #196

chris883773, 2019-11-17:
That is what i am looking for. 

> @chris883773 You might want to have a look at the linked PR #196

andi34, 2019-11-17:
```
sudo -u www-data -s
cd /var/www/html
git pull https://github.com/rawbertp/photobooth rp-filter-pic
exit
```