andreknieriem/photobooth #197: Frage zur Liveview

mdkhome, 2019-11-15:
Hallo, ich nutze Raspi 3b + und habe das Installastionscript vom Wiki genommen um Photobooth zu installieren. Habe die PiCam angeschlossen und auch in der Raspi-Config freigegeben. Sie funktioniert.
Aber ich bekomme kein Vorschau angezeigt. Muss noch was extra geändert werden. Da in /admin ja der Hacken aktivert wurde für (Vorschau) gehe ich davon aus das es eigentlich laufen sollte.
Als System läuft Buster. Bin nicht sehr bewandert im Bereich Linux/Dabian deswegen hoffe ich auch einfache Ratschläge :)

Ich bedanke mich schonmal im Vorraus

andi34, 2019-11-15:
Wenn du auf Take Pic clickst siehst du eine Vorschau.

closed 2019-11-15 by andi34

mdkhome, 2019-11-15:
Genau das passiert nicht bekomme einen weißen Hintergrund und den countdown dann macht er das Foto auf meiner Canon via USB. Das Bild zeigt er mir auch an. Aber ein livebild nicht.

andi34, 2019-11-15:
Livepreview geht mit der Pi Cam

mdkhome, 2019-11-15:
Also versteh ich das richtig, Das Foto machen mit der Canon und gleichzeitig das Liveview von der Pi Cam geht nicht ?
Hatte am Anfang die Pi Cam laufen und da hatte ich auch kein Live Bild gesehen beim Auslösen

andi34, 2019-11-15:
Funktioniert bei mir einwandfrei.
https://github.com/andreknieriem/photobooth/wiki/FAQ#is-pi-camera-supported