andreknieriem/photobooth #198: Disable display while countdown

tobiashaas, 2019-11-15:
Hey zusammen,

Ist es möglich das Display, während des Countdowns auszuschalten?
Oder dass während des Countdowns ein schwarzes Bild angezeigt wird?

Ich bin am überlegen ob ich mir so einen LED Ring für meine Box kaufen soll.
[https://helge-suess.com/hardware/photobooth-led-ring-2/](url)
[https://youtu.be/q2i6Y5AKtBE](url)

Ich habe nämlich öfters die Erfahrung gemacht, dass die Leute immer ins Display, anstatt in das Objektiv schauen ... Damit würde ich den Blick automatisch auf das Objetiv ziehen ;-)

Liebe Grüße 
Tobias

obkram, 2019-11-15:
Kleiner Tip von mir, ich habe bei mir aus dem gleichen Grund den Text.. Cheeeeeeeese ... in „ bitte in die Kamera schauen“ umbenannt. ;)

tobiashaas, 2019-11-15:
Habe ich auch schon getestet! 
Die Leute lesen es nicht ... wenn was leuchtet oder sich etwas verändert, würde das sicher besser klappen ;-)

andi34, 2019-11-15:
Not tested, but should work
https://github.com/andi34/photobooth/commit/7bfb58f25eb6345077283fcd9d65b34bb4080787

[image]

andi34, 2019-11-15:
If you don't like the text for change keep "color" defined as is

tobiashaas, 2019-11-15:
Da hätte ich ja auch drauf kommen können ... 
danke dir, werde es testen sobald ich den LED Ring habe! 

Wird aber sicher funktionieren 😉

rawbertp, 2019-11-16:
Die Frage ist nur, wie steuerst du diesen LED-Controller (sinnvoll) an. Wäre eventuell eine sinnvolle Erweiterung, dass man bspw definierte GPIOs triggert...?!

obkram, 2019-11-16:
@tobiashaas habe mir das von dir verlinkte Projekt angeschaut. Find ich super. Allerdings ist die Anleitung auf Windows ausgelegt. Mit welchem System betreibst du denn deinen Photobooth?

tobiashaas, 2019-11-16:
@obkram das stimmt! Bin aber im Kontakt mit dem Hersteller 😉

> Hallo Tobias,

für den Raspberry Pi habe ich noch keine Anbindung gemacht. Die meisten Anwender lassen die Booth auf Windows PC laufen.
Der Ring ist aber über USB Serial ansprechbar. Das geht auch von einem Pi aus. Damit kann man deutlich mehr anstellen als über die GPIO.

Im Prinzip kann jeder der Kabel-Buzzer über die GPIO angeschlossen werden. Der Bluetooth Buzzer meldet sich als Tastatur oder Maus. Kann man aber auch als Bluetooth Serial machen.

andi34, 2019-11-16:
> (...) I updated my buzzer with a LED strip. It looks fantastic:
> take a look if you like:
> 
> taking collage:
> https://youtu.be/44J-a71J25E
> 
> taking pic:
> https://youtu.be/eKONHESbrMw
> 
> sorry for the terrible video format. i took those video for instagram :)

Aus den vorgestellten Projekten bei mir im Fork

obkram, 2019-11-16:
@andi34 hatte ich gesehen. Reife Leistung von flighter18. Allerdings wenn ich das richtig gelesen hatte ist das von flighter18 nicht für das Wiki bestimmt. Vielleicht würde Tobias ja im Anschluss ein Wissen mit uns teilen ?!? ;)

andi34, 2019-11-16:
@obkram der BT Buzzer ist an sich halt keine günstige Sache, aber sehr edel. Werde denke in Zukunft auch so einen anschaffen.

flighter18, 2019-11-16:
> @andi34 hatte ich gesehen. Reife Leistung von flighter18. Allerdings wenn ich das richtig gelesen hatte ist das von flighter18 nicht für das Wiki bestimmt. Vielleicht würde Tobias ja im Anschluss ein Wissen mit uns teilen ?!? ;)

Hi @obkram was möchtest du denn über den Buzzer Wissen? 
Kann schon Auskunft geben, musst halt nur fragen  :)

obkram, 2019-11-16:
@flighter18 eine Frage hätte ich schon, hast du das auslösen der Count-down Sequenz in Verbindung mit dem Button nur auf dem Raspberry Pi Display realisiert oder hast du das auch auf deinem iPad auslösen können?

@andi34 Ich hatte mir mal Gedanken über eine etwas kostengünstigere Alternative gemacht. Mit den Funkrelais im angehängten Link ( oder alternativen ) sollte es normalerweise kein Problem sein.

https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.de%2Fulk%2Fitm%2F333300603608

Fernbedienung öffnen, zwei Kabel an den Button anlöten und dann auf den Potenzial freien Kontakt des Buzzers geben. Den Relais Kontakt der Gegenseite dann auf GPIO 24 legen. Damit sollte eigentlich ein kabelloser Buzzer unter 50 € realisierbar sein. Und den zweiten Button der Fernbedienung könnte man in Verbindung mit einem weiteren Script optional für das auslösen bei der Collage verwenden. Ich bin mir sicher dass man zu dem Skript Button.py auch parallel das Skript zum auslösen des LED Rings mit aufrufen kann, oder?

andi34, 2019-11-16:
Zur Zeit nutze ich einen ca. 10 Euro billigen Presenter zum Fernausloesen, funktioniert auch einwandfrei.
CSL - Presenter USB 2,4 GHz Fernbedienung für Präsentationen https://www.amazon.de/dp/B07T3BPL5H/ref=cm_sw_r_tw_dp_U_x_8s7ZDbK1QBYHA

tobiashaas, 2019-11-16:
> @andi34 hatte ich gesehen. Reife Leistung von flighter18. Allerdings wenn ich das richtig gelesen hatte ist das von flighter18 nicht für das Wiki bestimmt. Vielleicht würde Tobias ja im Anschluss ein Wissen mit uns teilen ?!? ;)

selbstverständlich ;-)

rawbertp, 2019-11-16:
@flighter18 super Sache! Wie synct sich denn der Buzzer mit PB (Countdown, Pic taken, ...)?

obkram, 2019-11-16:
@andi34 flighter18 schrieb mal etwas von einer Differenzierung im Script zwischen Short Press und Long Press zur Auslösung eines Fotos oder einer Collage. Habt ihr da schon was umgesetzt?

andi34, 2019-11-16:
> @andi34 flighter18 schrieb mal etwas von einer Differenzierung im Script zwischen Short Press und Long Press zur Auslösung eines Fotos oder einer Collage. Habt ihr da schon was umgesetzt?

Nein.

flighter18, 2019-11-16:
> @flighter18 dein Buzzer ist schon Königsklasse. Eine Frage hätte ich schon, hast du das auslösen der Count-down Sequenz in Verbindung mit dem Button nur auf dem Raspberry Pi Display realisiert oder hast du das auch auf deinem iPad auslösen können?

Buzzer und LED Strip ist Systemunabhängig. Somit funktioniert beides auch mit dem einem Pad.
Wichtig zu wissen ist noch, dass die beiden Systeme Buzzer und LED Strip voneinander unabhängig sind. :)

Der Buzzer verbindet sich per Bluetooth mit dem PI oder iPad und wird  schliesslich als Tastatur erkannt. 

Der LED Strip funktioniert per WLAN.  Der Webserver, iPad muss dann Zugriff auf dieses WLAN haben. 
Bei mir ist es so, dass der PI ein WLAN macht und das iPad + LED Strip sich dann verbinden.
Sobald der Buzzer gedrückt wird, macht der Photobooth auf dem PI eine HTTP Abfrage auf den LED Strip und löst dann das Licht aus.

Countdown:
```
       if (start == 5) {
               console.log('five');
               var request = new XMLHttpRequest();
               request.open(### "GET","http://10.0.0.2/BACK");
               request.send();
       }
       if (start == 2) {
               console.log('two');
               var request = new XMLHttpRequest();
               request.open(### "GET","http://10.0.0.2/TWO");
               request.send();
       } 
```
Regenbogen sobald das fertige Foto angezeigt wird
```
       if (data.error) {
                   public.errorPic(data);
               } else {
                       if (photoStyle === 'collage') {
                               console.log('COLLAGE LED');
                               var request = new XMLHttpRequest();
                               request.open(### "GET","http://10.0.0.2/COLLAGE");
                               request.send();
                       }else{
                               console.log('PIC LED');
                               var request = new XMLHttpRequest();
                               request.open(### "GET","http://10.0.0.2/PIC");
                               request.send();
                       };
                       public.renderPic(data.file);
               }
```

Für die Steuerung des LED Stripes habe ich diesem Controller gekauft:
https://de.aliexpress.com/item/32651747570.html?spm=a2g0s.9042311.0.0.452a4c4dVu4Uyf

Das Controller Gehäuse habe ich mit Autodesk Fusion 360 entworfen und dann mit dem 3D Printer ausgedruckt.

[image]
[image]

Mittlerweile habe ich auch eine neue Box gemacht 😄 
[image]

flighter18, 2019-11-16:
> @flighter18 super Sache! Wie synct sich denn der Buzzer mit PB (Countdown, Pic taken, ...)?

Siehe mein vorherige Erklärung. 😄

flighter18, 2019-11-16:
> @andi34 flighter18 schrieb mal etwas von einer Differenzierung im Script zwischen Short Press und Long Press zur Auslösung eines Fotos oder einer Collage. Habt ihr da schon was umgesetzt?

Das Board, welches ich verwende ist und programmieren tue ich es mit Arduino. 
https://www.play-zone.ch/de/adafruit-feather-32u4-bluefruit-le.html

Für das Board gibt es schon Bibliotheken, welche ich verwendet habe. 
Im Script selbst unterscheide ich nur zwischen:

``` 
singleclick()                           
longclick()

```

https://learn.adafruit.com/introducing-the-adafruit-bluefruit-spi-breakout/hidkeyboard

obkram, 2019-11-16:
@flighter18 Danke das du dein Projekt mit uns teilst. Schöne Zusammenfassung. Klingt schlüssig soweit. Muss man sich etwas rein arbeiten.

flighter18, 2019-11-16:
@obkram kein Problem. Falls du noch mehr Infos brauchst, einfach schreiben. 😄 
Ah und du hast recht, muss sich ein wenig rein arbeiten! Das Ganze macht aber sau spass 👍 

Noch ein interessanter Aspekt an meinem Photobooth, ist, dass alle Daten direkt auf eine Cloud hochgeladen werden. 😸 
Dafür habe bei mir eine Nextcloud "https://nextcloud.com" installiert. Die Fotos werden dann per UMTS und WebDav direkt auf die Cloud in einen Ordner geladen. 
Den Ordner wird dann schliesslich per Link geshared. Aus diesem Link generiere ich dann ein QR-Code, welche die Leute scannen können. Somit haben sie direkten Zugriff auf die Fotos und müssen nicht über das WLAN des Photobooths gehen.

Die UMTS Verbindung verwende ich dann auch für den Support, wenn mal etwas nicht geht:
Sobald man die Box startet wird nämlich eine SSH Verbindung inkl Portforwarding "https://www.ssh.com/ssh/tunneling/example" zu mir ins Heimnetz gemacht. Somit habe ich Zugriff auf die PI Shell, CUPS und auch auf den WEB GUI und zwar egal wo die Box steht. 
Natürlich muss ein 3G bzw 4G Netzwerk vorhanden sein.

rawbertp, 2019-11-16:
@flighter18 Very nice👍 Danke für die Infos.

Next-/Owncloud-Anbindung wäre btw ein cooles Feature IMO! Magst du einen PR machen? 😎

rawbertp, 2019-11-16:
@flighter18 noch eine Frage: kannst du den Source Code zur Ansteuerung des LED Stripes zur Verfügung stellen?

flighter18, 2019-11-17:
> @flighter18 Very nice👍 Danke für die Infos.
> 
> Next-/Owncloud-Anbindung wäre btw ein cooles Feature IMO! Magst du einen PR machen? 😎

Was meinst du mit PR? 😸 

Hier noch der Code:
[LED_STRIP.zip](https://github.com/andreknieriem/photobooth/files/3855692/LED_STRIP.zip)

flighter18, 2019-11-17:
@rawbertp noch etwas wichtiges
sollest du nicht die gleiche Anzahl LED wie ich haben, musst du den Code anpasse :)

obkram, 2019-11-17:
@flighter18 würdest du evtl. auch deinen Code für den Buzzer weitergeben? :)

andi34, 2019-11-17:
Jemand gute Erfahrungen mit einem Touch Screen 10-12 Zoll für den Pi?

tobiashaas, 2019-11-17:
@andi34 Ich besitze dieses hier 
https://eckstein-shop.de/101-inch-HDMI-LCD-B-with-case

Bin super zufrieden damit

flighter18, 2019-11-18:
> @flighter18 würdest du evtl. auch deinen Code für den Buzzer weitergeben? :)

@andi34 gäbe es die Möglichkeit, den Buzzer-/LED Code im Wiki oder im Repo unterzubringen? Dann würden alle Leute davon profitieren! Sonst profitieren ja nur die Leute, die diesen Issue durchlesen und Deutsch verstehen, was ich nicht so fair finde :smile:

flighter18, 2019-11-18:
> Jemand gute Erfahrungen mit einem Touch Screen 10-12 Zoll für den Pi zum Direktanschluss?

Ich verwende auch einen WaveShare Display, wie tobiashaas und bin auch sehr zufrieden.
Lustig ist, dass die Displays in der Schweiz günstiger sind: 
https://www.pi-shop.ch/waveshare-display-10-1inch-hdmi-lcd-b-with-case-1280-800-ips

Im neuen Photobooth habe ich dieses Display eingebaut:
https://www.pi-shop.ch/13-3inch-hdmi-lcd-h-v2-1920x1080

andi34, 2019-11-18:
> @andi34 gäbe es die Möglichkeit, den Buzzer-/LED Code im Wiki oder im Repo unterzubringen? Dann würden alle Leute davon profitieren! Sonst profitieren ja nur die Leute, die diesen Issue durchlesen [...]

Klar, schreib es einfach hier rein in englisch passend als Anleitung, dann kann ich's einarbeiten.

andi34, 2019-11-18:
Wenn du willst kannst du auch eine eigene Repo machen, Jenachdem wie viel Code es ist. Z.B. photobooth-buzzer-led oder ähnlich und pusht da alles hin. Dann kann man es verlinken.

flighter18, 2019-11-24:
> Wenn du willst kannst du auch eine eigene Repo machen, Jenachdem wie viel Code es ist. Z.B. photobooth-buzzer-led oder ähnlich und pusht da alles hin. Dann kann man es verlinken.

@andi34 Sorry hatte diese Woche wenig Zeit.
Guten Idee. es ist zwar nicht viel Code aber ich denke, ein eigenes Repo wäre nicht schlecht. 

Ich hoffe nächste Woche habe ich ein wenig mehr Zeit.

flighter18, 2019-12-04:
Hallo zusammen

Habe nun endlich geschafft die Scripts in ein Repo zu laden:
https://github.com/flighter18/photobooth-buzzer-led

Mit der Anleitung habe ich noch nicht angefangen 😅

andi34, 2019-12-09:
https://github.com/andi34/photobooth/commit/07f0f10f0f339c38a4b0880ebf6d475b3cbec7a1

Neue config:
Schrift ist weiss, Hintergrund dunkel.

Habe einen Kommentar hinterlassen im commit wie sie Schrift ausgeblendet wird (also auch keine Fehler Benachrichtigung!)

andi34, 2020-01-01:
[image]

ich-dev, 2020-01-02:
Muss ich um einen Buzzer an den GPIO anzuschliessen ein wiederstand anlöten oder geht das ohne wiederstand?

closed 2020-01-15 by andi34

weili87, 2020-07-08:
> > @flighter18 dein Buzzer ist schon Königsklasse. Eine Frage hätte ich schon, hast du das auslösen der Count-down Sequenz in Verbindung mit dem Button nur auf dem Raspberry Pi Display realisiert oder hast du das auch auf deinem iPad auslösen können?
> 
> Buzzer und LED Strip ist Systemunabhängig. Somit funktioniert beides auch mit dem einem Pad.
> Wichtig zu wissen ist noch, dass die beiden Systeme Buzzer und LED Strip voneinander unabhängig sind. :)
> 
> Der Buzzer verbindet sich per Bluetooth mit dem PI oder iPad und wird schliesslich als Tastatur erkannt.
> 
> Der LED Strip funktioniert per WLAN. Der Webserver, iPad muss dann Zugriff auf dieses WLAN haben.
> Bei mir ist es so, dass der PI ein WLAN macht und das iPad + LED Strip sich dann verbinden.
> Sobald der Buzzer gedrückt wird, macht der Photobooth auf dem PI eine HTTP Abfrage auf den LED Strip und löst dann das Licht aus.
> 
> Countdown:
> 
> ```
>        if (start == 5) {
>                console.log('five');
>                var request = new XMLHttpRequest();
>                request.open(### "GET","http://10.0.0.2/BACK");
>                request.send();
>        }
>        if (start == 2) {
>                console.log('two');
>                var request = new XMLHttpRequest();
>                request.open(### "GET","http://10.0.0.2/TWO");
>                request.send();
>        } 
> ```
> 
> Regenbogen sobald das fertige Foto angezeigt wird
> 
> ```
>        if (data.error) {
>                    public.errorPic(data);
>                } else {
>                        if (photoStyle === 'collage') {
>                                console.log('COLLAGE LED');
>                                var request = new XMLHttpRequest();
>                                request.open(### "GET","http://10.0.0.2/COLLAGE");
>                                request.send();
>                        }else{
>                                console.log('PIC LED');
>                                var request = new XMLHttpRequest();
>                                request.open(### "GET","http://10.0.0.2/PIC");
>                                request.send();
>                        };
>                        public.renderPic(data.file);
>                }
> ```
> 
> Für die Steuerung des LED Stripes habe ich diesem Controller gekauft:
> https://de.aliexpress.com/item/32651747570.html?spm=a2g0s.9042311.0.0.452a4c4dVu4Uyf
> 
> Das Controller Gehäuse habe ich mit Autodesk Fusion 360 entworfen und dann mit dem 3D Printer ausgedruckt.
> 
> [image]
> [image]
> 
> Mittlerweile habe ich auch eine neue Box gemacht smile
> [image]

Hallo zusammen,

erstmals vielen Dank für die Tolle Arbeit, ich finde das PhotoBooth echt klasse.

aber zu mein Problem, ich bin Neuling und hab noch nicht so viel Ahnung davon, bin gerade dabei es zu verstehen :-)
Ich versuche schon ein paar tage die Codezeilen einzufügen wo sie hin gehören. Ich finde aber nicht die richtige Datei und den richtigen Platz dazu. Kann mir von eich wer Helfen wo ich die Codezeilen einfügen muss?

> Countdown:
>
> ```
>        if (start == 5) {
>                console.log('five');
>                var request = new XMLHttpRequest();
>                request.open(### [...]

Liebe Grüße
Markus