andreknieriem/photobooth #207: Backward compatibility: iPad2 / iOS 9.3.5

jacques42, 2019-12-10:
Hi - first, many thanks for this great solution. Thanks a million for creating it and keep it up please, it is excellent - I'm in the process of to build my own photobox and yours is really what I am looking for. 

I would like to use my (reasonably old) iPad2 in the photobox, with Photobooth on it. But found that the current master (2.1.0-alpha)  is not working on iOS 9.3.5 (latest for iPad2). 

Now - I am not at all a Javascript coder but thought I'd give it a try as Photobooth really is a great solution. So having a closer look I think I found three minor adjustments, which together seems make the code work (also) on iOS9 / Safari 9

First, and a most general thing, Javascript ES6 (https://webkit.org/blog/4054/es6-in-webkit/) is not supported in iOS 9 / Safari 9. As a consequence
1. Do not use `let`, use `var`instead
2. Do not use Arrow Functions ` () => {....} `- use standard function notation `function() { ... }`

Second, in the Gallery I noticed that when in Photoswipe the feature `tapToToggleControls` (default: true) will hide / show the controls (as designed) but also prevent buttons from working. So i.e. the download button or the qr code button did not work, likely the buttons did not receive a `click` or `touchstart` event. 
```
// Tap should toggle visibility of controls
tapToToggleControls: true,
```
I decided not going deeper in Photoswipe but to switch off this feature in the Photobooth code. Done that in function `openPhotoSwipe()`, when creating the `PhotoSwipe` object and for me it works well (actually I prefer the controls visible all the time).

I realize this is a backward compatibiliy topic and maybe not high on the priority list. And this is fine. Yet having done the work, let me share for your possible consideration. I'll add a pull request shortly. 

Best,
Jacques42

andi34, 2019-12-11:
Hey and thanks for your feedback and contribution.
https://github.com/andreknieriem/photobooth/pull/191 is merged into master branch and should already have fixed the issue with not responding buttons inside the gallery on older iOS versions.

You're sure you're on latest master branch?

From webkit.org it seems like ES6 is supported in iOS 9 (link given above).

[image]

Best regards

Andi

jacques42, 2019-12-11:
Hi - I'm on version 2.0.2 which is the latest in the master branch (?) - i had looked at file `package.json `which I should not have, sorry for the confusion. Changes from #191 are applied in my local copy, thanks for pointing it out. 

Version check:
```
root@photoboothpi:/var/www/html/photobooth# git describe --tags '`git rev-list --tags --max-count=1'`
v2.0.2
```

`let` and Arrow Functions definitely don't work on my iPad2 iOS9.3.5 - using any of the two immediately crashes the code in Safari.

Now the fun part: Today, I can't reproduce the Photoswipe controls behavior that I witnessed before. I took a fresh install, fixed let and arrow functions but no way I was able to reproduce the Photoswipe controls toggle (mis-)behavior. 

This feels a bit silly now :-) as something was wrong but now I don't know and worse I can't reproduce.  Maybe browser cache or something - idk. So I guess I have to live with this and at least the whole gave me some great learnings (getting started only !) on Javascript, PHP and CSS.

Let me close this one here, thanks for taking the time to look at it anyways. 

Best,
Jacques42

closed 2019-12-11 by jacques42

andi34, 2019-12-11:
@jacques42 you've tested latest webkit nightly which supports let and arrow function?
https://webkit.org/blog/29/nightly-builds/

jacques42, 2019-12-11:
@andi34 Indeed arrow functions are listed as functional in ES6, but other sources seem to indicate they are only supported on iOS as of v10: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Functions/Arrow_functions and https://caniuse.com/arrow-functions

iOS 9.3.5 uses WebKit v601.1.46 as per user agent string, so I am not sure about this exact build capabilities. 

I really tried but I feel somewhat stupid now as I am not clear on how test another webkit build on my iPad2. I don't think there is an obvious, easy way to do so. I did google and Apple Developers suggests to use XCode to create a iOS simulation environment. Not sure really if this would be the way forward, I decided to not go deeper. 

Now please - it feels like I do and I don't want to waste your time by this topic. iOS 9.3.5 is water under the bridge definitely and nothing to be concerned about in the bigger picture. I definitely appreciate your time on the matter but I think I will keep it at this point. My work-around seems ok to match my use-case and essentially I am very happy that I'll be able to use the Photobooth software at it's best. :-)

Thanks,
Jacques42