andreknieriem/photobooth #212: Count printed pictures

danielovic295, 2019-12-25:
**Is your feature request related to a problem? Please describe.**
It would be nice to have a counter which shows me the amounts of printed pictures. As me and many other use a DNP printer, which has no display showing the remaining possible prints on the paper roll, you always run into danger, that the roll with go empty while having an event. 

**Describe the solution you'd like**
A simple counter, maybe in the admin page (has to be resettable).

**Describe alternatives you've considered**
I always took the amount of files in the /data/print/ directory, but on that way I can't count the prints which got printed several times. 

**Additional context**
A possible solution could be to count the executions of the print.php file, but myself is not able to code something like that.

flighter18, 2019-12-25:
HI @danielovic295 

I did something simalery. My goal was a webpage with all the photobooth statistics. This is the result:

**Main page**
[image]

**All taked Photos** 
[image]

I use a PHP API and modified the takePic.php and print.php file. Now every time someone prints a picture or makes a photo a curl request will be send to my server. My server takes the request iand saves them in a Database. Finally my website  evaluate the database and shows all the metrics in a nice way.

maybe that's a little over the top but let me know, if you want more infos. 😄

danielovic295, 2019-12-26:
Hi @flighter18,

this is brilliant, but for the live monitoring there is connection to the internet required, isn't it? 

Besides that, I think I am not able to implement that into my system.

What resources do you need for that system? 

But I think it is not absolutey necessary to have a live monitoring (but it's very nice tho :D), a simply offline tracking would be absolutely sufficient.

BR,
danielovic

andi34, 2019-12-26:
Here's an example how to implement a counter:
https://github.com/andreknieriem/photobooth/commit/ab6cc50d5571cdd9245d7c5b0801484f284872b7#diff-1463ee9d4350b7e153b2bf7b62a8f151

flighter18, 2019-12-27:
> Hi @flighter18,
>
> this is brilliant, but for the live monitoring there is connection to the internet required, isn't it?
>
> Besides that, I think I am not able to implement that into my system.
>
> What [...]

Actually you don't need an internet connection. you could do it on the photobooth hardware. 
The only resource you additional need is a Database.

maybe you could tryout andi34's implementation. 😄

andi34, 2019-12-27:
It's not my implementation, but a hint on how it could work and how it could be implemented.

flighter18, 2019-12-27:
@andi34 sorry for the misunderstanding

ich-dev, 2020-01-05:
@flighter18 könntest du mich per email anschreiben?
Wohne in deiner nähe (baselland) bojan.rauch@gmail.com

flacoonb, 2020-02-29:
Hi @flighter18 
that's exactly what I'm looking for. Looks great! Can you provide your code? 
Thank you very much.

flighter18, 2020-03-02:
hi @flacoonb 

You will find the code here:
https://github.com/flighter18/photobooth-buzzer-led

flacoonb, 2020-03-04:
thx @flighter18 
but the link is for buzzer and not for the count printed pictures?

flighter18, 2020-03-06:
ou sorry wrong theme :)
currently the code isn't on any git repo. 😃

flacoonb, 2020-03-06:
😉 
all right, that would be great! I'm sure there are even more interested.
Thanks a lot already!

flacoonb, 2020-03-20:
@flighter18 
what about the repo? Thx

flighter18, 2020-03-21:
@flacoonb 
https://github.com/flighter18/photobooth-statistic