andreknieriem/photobooth #213: Croma Key

Joeol, 2019-12-28:
Hallo, ich habe mir mit der Photobooth Software einen Raspberry Pi 4 4 GB eingerichtet. Der Raspberry macht alles, den Wlan AP, den Webserver und das Frontend.  An dem Raspberry habe ich einen 15" Touchscreen Monitor. Es läuft alles top und auch recht zügig, selbst das Livebild über die PI Camera, nur beim Croma Key habe ich das Problem, dass das Bild nicht gespeichert wird, sondern der Screen immer auf "speichern" steht. Ich habe im Chrom schon den entsprechenden Flag, wie beschrieben, disabled, jedoch will es einfach nicht funktionieren. Vielleicht hat ja jemand einen Hinweis für mich, was ich falsch mache.
Vg
Jörg

andi34, 2019-12-30:
Mal mit F12 in die Console Des browsers geschaut?

Joeol, 2019-12-30:
Ja, schon, leider kann ich das nicht interpretieren, bin mehr der Schrauber als der Programmierer.

andi34, 2019-12-30:
Schreit er denn mit irgendeinem Fehler?
Kannst auch Screenshot hier Posten

andi34, 2019-12-30:
Funktioniert es mit einem Foto dev mode?

Wenn ja, schraub Mal die Bildgröße in der kamera runter.

andi34, 2019-12-30:
Ok, kann ich nachvollziehen.

[image]

andi34, 2019-12-30:
https://datanextsolutions.com/blog/how-to-fix-nginx-413-request-entity-too-large-error/

Joeol, 2019-12-30:
Ich bekomme in "Console" die Fehler
Failed to load resource: the server responded with a status  /favicon.ico:1
 of 404 (not Found)
Failed to Load recource: the Server responded /api/chromakeying/save.php:1 
with a status of 413 (request Entity Too Large)
Die Bildgrösse habe ich mal runtergstellt, dass Ergebnis ist das Selbe.

Joeol, 2019-12-30:
Ok, schaue ich mir an, probiere ich aus und melde dann Erfolg, vielen Dank für die Hilfe.

andi34, 2019-12-30:
Funktioniert. Einfach dem Link folgen und entsprechend die nginx Config ändern.

[image]

closed 2019-12-30 by andi34

Joeol, 2019-12-30:
Läuft, vielen Dank für die Hilfe.