andreknieriem/photobooth #216: Simple way to set Exposure of RPi Cam

LDFOTO, 2020-01-07:
Preview of RPi cam is too dark

Maybe it is possible to set it to auto exposure or add a method to set parameters for given lighting situation.

andi34, 2020-01-07:
It's a stream using the device cam and there's nothing changed by Photobooth on default brightness/exposure. It's using default values, not sure if you can adjust them.

LDFOTO, 2020-01-07:
Ok. Isn't there any configuration file for cheese?

andi34, 2020-01-07:
The cheese text ist added after the countdown, time "cheese" is visible can be adjusted.

LDFOTO, 2020-01-07:
How do I adjust exposure?

andi34, 2020-01-07:
Settings for the stream:
https://github.com/andreknieriem/photobooth/blob/18c07f2/resources/js/core.js#L11-L17

Start and stop the stream:
https://github.com/andreknieriem/photobooth/blob/18c07f2/resources/js/core.js#L97-L126

This gets called here to start the stream once you click on take picture button:
https://github.com/andreknieriem/photobooth/blob/18c07f2/resources/js/core.js#L137

Stopped here https://github.com/andreknieriem/photobooth/blob/18c07f2/resources/js/core.js#L173 right before the picture is taken.

Cheese text is added here https://github.com/andreknieriem/photobooth/blob/18c07f2/resources/js/core.js#L155

LDFOTO, 2020-01-07:
I went to Cheese project page. Can't find anything on exposure settings.

andi34, 2020-01-07:
You could try changing https://github.com/andreknieriem/photobooth/blob/master/resources/sass/style.scss#L95

From `background: rgba(255,255,255,0.9);` to `background: #ffffff;`

And see if it makes a difference. Make sure running `yarn build` after changing the sccs file

LDFOTO, 2020-01-07:
thx. I'll try on Friday and will report

andi34, 2020-01-07:
👍 that's the only thing I can think of which could make a difference, else there's no way I know.

LDFOTO, 2020-01-10:
I can see no change. Is it possible to use an alternative to cheese?

andi34, 2020-01-10:
There's nothing changed by Photobooth besides the opacity from cheese which could make the preview a bit dark.

You'll have to dig around and see if you can change it inside Raspbian by default then.

You can change the cheese text inside the language file. Not sure what you like to change.

LDFOTO, 2020-01-10:
First I thought Raspicam is controlled by raspivivid which allows several settings on exposure. Can't find anything on how cheese is controlling exposure. Seems to be a fixed setting.

LDFOTO, 2020-01-10:
Raspicam seems to have no auto exposure feature by itself. Specification says "controlled by ISP"

andi34, 2020-01-10:
We're only accessing the device cam in the Webinterface, therefore Default Parameter from PiCamera library are used and I don't know a way you could change it.

LDFOTO, 2020-01-10:
There seems to be no way to solve this. thx

closed 2020-01-10 by LDFOTO

andi34, 2020-01-10:
You could try adding `filter:brightness(400%);` in the style.scss for the video

https://github.com/andreknieriem/photobooth/blob/master/resources/sass/style.scss#L339

Change:
```
#video {
  position: absolute;
  top: 50%;
  margin-top: -240px;
  margin-left: -360px;
  left: 50%;
}
```
To
```
#video {
  position: absolute;
  top: 50%;
  margin-top: -240px;
  margin-left: -360px;
  left: 50%;
  filter:brightness(400%);
}
```