andreknieriem/photobooth #217: Trigger photobooth via HTTP request

maxmlr, 2020-01-14:
First of all, great project and amazing work!

I am wondering which would be the best way to realize a remote trigger button. That could be a simple wifi button sending a GET/POST request (i.e. to /trigger) along with a secret to trigger the photobooth.

I understand that the solution for the GPIO is emulating a key press using the key defined in the configuration file. This could be also done here but I think i would not be a nice solution.

As I plan to implement this functionality, in your opinion, what would be the most fitting way to approach this in the current implementation?

rawbertp, 2020-01-16:
I doubt that'll be easy to implement. The browser would have to listen and respond to this API call and that's simply not the way it's currently designed for (neither the front- nor the backend).

sualko, 2020-01-16:
You can just call the same urls as the javascript. This means `api/takePic.php` and `api/applyEffects.php` with the appropriate options. You find this "documented" in the js file :wink:

closed 2020-01-16 by sualko

rawbertp, 2020-01-16:
Yes, you could do that. But how would the browser displaying the picture know about that action being triggered from somewhere else? So you could trigger the photo but the browser "in the booth" would just display nothing - I doubt that's the expected behaviour. ;o)

sualko, 2020-01-16:
@rawbertp you are absolutely right, it's not that easy.

@maxmlr you could write a simply python script which creates a http server and triggers a key press on some url. It's a bit of a hack, but it's probably the fastest way.

maxmlr, 2020-01-17:
Thanks for the comments - clearly it's not straight forward, but I think it would be a valuable feature.
@sualko I will think about a lightweight solution for this and submit a pull request.

jacques42, 2020-02-01:
@maxmlr Same thing here, and I coded a solution, using socket.io . You may want to have a look https://github.com/jacques42/photobooth/tree/socketio-HW-button

It's fully integrated in photobooth (check the admin settings), yet fairly fresh and not really tested for robustness on other platforms. However it works like a charme in my setup, which is RaspberryPi as webserver, iPad2 as front-end. For the buzzer I have a 3d-printed button connected to the respective GPIO, which will connect to ground (PULL_DOWN) when triggered. 

In case you do not want to use a hardware button on the Pi, any socket.io client can connect and trigger from remote. 

I'd be interested in feedback and others testing, to make it more robust, in case people are interested.