andreknieriem/photobooth #219: Frage: Live-Wallpaper

PatXoXo, 2020-01-24:
Hallo zusammen. 
Ich habe eine Frage zum Live-Wallpaper. 
Bei mir läuft die Photobooth auf einem Pi 3 und zusätzlich nutze ich ein Windows-Tablet zum Auslösen & Bilder sehen. 

Ich habe versucht jetzt einen Live-Stream als Background einzurichten. 
Dazu habe ich mir auf dem Windows-Tablet Yawcam installiert, weil ich die integrierte Cam nutzen wollte. 
Ich kann diese Kamera auch über einen normalen Browser erreichen, nur leider bekomme ich diesen Stream nicht als Hintergrund gesetzt. 
Die passende Adresse habe ich im Admin-Panel gesetzt, ich sehe leider nichts. 
Adresse ist eingetragen im Format: url(http://192.168.178.66:8081/). 

Hoffe ihr könnt mir kurz helfen, danke & Gruß

andi34, 2020-01-24:
Für die Vorschau während des Countdowns musst du die Booth per HTTPS aufrufen oder eine Ausnahme im Browser hinterlegen, sonst darf er nicht auf die Gerätekamera zugreifen.
https://medium.com/@Carmichaelize/enabling-the-microphone-camera-in-chrome-for-local-unsecure-origins-9c90c3149339

Siehe auch: https://github.com/andreknieriem/photobooth/issues/202

Für den grundsätzlichen Stream als Hintergrund auf der Startseite, sofern der Aufruf per Browser auf den Stream funktioniert:
- url nochmal kontrollieren (Leerzeichen irgendwo oder so?)
- ggf Browser cache löschen oder über ein privates Fenster öffnen

Kannst auch gerne deine my.config.inc.php Posten, Logindaten für Email vorher rauslöschen falls eingestellt!

PatXoXo, 2020-01-24:
Hey, vielen dank schon mal für die Antwort!
Jetzt wo ich die Ausnahme erstellt habe sehe ich auch die Vorschau aus der Kamera :). 

Ich hab nur noch 2 Fragen dazu
a) Die Aufnahme fängt erst bei Sekunde 3/5 an. Ist das eine Einstellungssache oder liegt das am verzögerten Ansprechen der Kamera? 
b) Würde es auch funktionieren eine USB-Cam an den Laptop anzuschließen und von dort die Vorschau zu erzeugen?

andi34, 2020-02-19:
A) die Kamera braucht natürlich einen Moment den Stream zu erzeugen

B) ggf., Kommt darauf an wie sie am Ende vom OS behandelt wird - kann funktionieren muss aber nicht.

closed 2020-02-19 by andi34