andreknieriem/photobooth #222: Picture from device cam is not saved, instead random picture is shown

irqnet, 2020-02-01:
**Describe the bug**
Use device camera on android phone. Preview is shown, but picture is not displayed and stored (random image is shown)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'photoboot URL with https' on mobile device
2. Click on 'take picture'
3. Preview shows live picture from device cam
4. Picture is taken and random preview picture appears

**Expected behavior**
Picture from device cam should be stored on photobooth gallery

**Screenshots**
1. https://i.imgur.com/UVSQr5x.jpg
2. https://i.imgur.com/6tLKjAx.jpg

**Environment (please complete the following information):**
 - OS: Android (Xiaomi Mi9 with MIUI 11.0.5)
 - Browser Chrome

**Photobooth**
- Did it work before? no
- Which version of Photobooth are you using? 2.1.0

andi34, 2020-02-01:
How does your my.config.inc.php look like? Random image is used if dev mode is enabled. Make sure you have not enabled it.

irqnet, 2020-02-01:
Dev Mode was enabled - it's working now

closed 2020-02-01 by irqnet