andreknieriem/photobooth #223: Fehler wenn die Camera nicht an ist.

Joeol, 2020-02-10:
Hallo, ich benutze die aktuelle V.2.10 und habe festgestellt, dass wenn man vergißt, die SLR Camera anzumachen, beim Drücken auf den "Foto erstellen" Knopf sich der Browser so aufhängt, dass man nichts mehr bedienen kann, wenn der Browser im Kioskmode läuft. Es ist nicht relevant, ob die Firefox, oder Chrome nehme.

Klar gibt es keinen Sinn die Kamera nicht anzuhaben, aber ist mir nun schon 2 Mal passiert und dann geht im Browser nichts mehr. Vielleicht könnte man das ja irgendwie abfangen. In der vorherigen Version konnte man die Seite dann neu laden.

VG
Joe

`GET http://localhost:8081/ net::ERR_CONNECTION_REFUSED
core.js:114 [Deprecation] getUserMedia() no longer works on insecure origins. To use this feature, you should consider switching your application to a secure origin, such as HTTPS. See https://goo.gl/rStTGz for more details.
public.startVideo @ core.js:114
public.thrill @ core.js:142
(anonymous) @ core.js:510
dispatch @ jquery.min.js:2
v.handle @ jquery.min.js:2
core.js:121 Could not get user media:  DOMException
core.js:199 took picture {success: "image", file: "20200210_194514.jpg"}
core.js:284 picture processed {error: "Could not read jpeg file. Are you taking raws?"}`

andi34, 2020-02-10:
Fehler kommt von https://github.com/andreknieriem/photobooth/blob/stable2/api/applyEffects.php#L49

Aber eigentlich sollte er hier schon abbrechen
https://github.com/andreknieriem/photobooth/blob/stable2/api/takePic.php#L41
(Gphoto2 kann das Bild ja nicht aufnehmen).

Und eigentlich sollte durch https://github.com/andreknieriem/photobooth/commit/0ed52d9f72fd108040ca2f357dea31f47549e4e7 auch spätestens hier abgebrochen werden und ein "Seite neu laden" Button angezeigt werden zusammen mit der Fehlermeldung (sofern nicht deaktiviert).

Die v2.1.0 sauber installiert?
Nur im Kiosk Mode?

Joeol, 2020-02-10:
Hallo, ich habe ein Upgrade von der 2.0 gemacht. Das Problem tritt auch ohne Kioskmode auf. Werde es einmal ganz frisch installieren und mich dann noch mal melden, wenn das Problem noch besteht.

Joeol, 2020-02-10:
Stimmt, nach der frischen Installation spring das Programm wieder auf den Startbildschirm, wenn man neu laden drückt.

Vielen Dank für de Hilfe
Joe

andi34, 2020-02-10:
Sehr gut :)

closed 2020-02-10 by andi34