andreknieriem/photobooth #227: Preview Image dont fits in Screen

WeAppU, 2020-02-26:
At First, FANTASTIC Work and Thank you very much for that...

**Describe the bug**
After taking a Picture the Preview Picture is Bigger then the Screen.
Where can i ajust that the Preview Picture fits 100% in Widght and Hight?

**Environment (please complete the following information):**
 - OS: Raspberry 4, 4GB
 - Browser: Chromium
 - Version: Last

**Photobooth**
- Did it work before? No
- Which version of Photobooth are you using? Newest

Wild-Bird, 2020-02-29:
I also don't like the Result Screen.
I changed one point in /var/www/html/resources/css/style.css that i can see the whole Picture.

Original:
`#result {
  width: 100%;
  background: center center no-repeat;
  background-size: cover;
  z-index: 2; }`

My quick and dirty solution:
`#result {
  width: 100%;
  background: center center no-repeat;
  background-size: contain;
  z-index: 2; }`

Not perfect, but you can see the whole Picture then. 
Your Picture can only fit 100% in Width and High when the ratio of your browserwindow is exactly the same as the Picture.   

Is there a better solution, please let me know.

andi34, 2020-03-01:
@Wild-Bird seems to work fine for me too. Thanks for sharing your solution! If someone else likes to test too: [photobooth-dev-SNAPSHOT.20200301083744205.zip](https://github.com/andreknieriem/photobooth/files/4271531/photobooth-dev-SNAPSHOT.20200301083744205.zip)

WeAppU, 2020-03-01:
> 
> 
> @Wild-Bird seems to work fine for me too. Thanks for sharing your solution! If someone else likes to test too: [photobooth.zip](https://github.com/andi34/photobooth/releases/download/v2.2.1/photobooth-2.2.1.zip)
> 
> Edit:
> Habe das zip nochmal neu hochgeladen weil ein typo drin war.

Kann ich das einfach "drüberbügeln"?

andi34, 2020-03-01:
> Kann ich das einfach "drüberbügeln"?

Wüsste nicht warum nicht. Aber mach dir gerne vorher ein Backup von deinem jetzigen Setup. Ggf. Die login.php und logout.php nur noch entfernen, da ich diese in den Unterordner login gepackt habe damit der Aufruf über [localhost/login](localhost/login) und nicht [localhost/login.php](localhost/login.php) läuft.

andi34, 2022-10-04:
Can be defined via Adminpanel in Photobooth v4
https://photoboothproject.github.io/Changelog

closed 2022-10-04 by andi34